cyberbotics/webots #1236: External Controller for ML Cloud Training ?

MentalGear, 2020-01-06:
Is it currently possible using Webots with the external modus to train robots controled by ML via the cloud (like google colab) ?

Cloud Training is an important asset for ML, and it would be great if Webots would provide some basics guides to ensure easy usage and/or, if not possible yet, implement cloud training functions assure it stays compatible & competitive with current developments in the field.

(As an aside, some basic guides on ML implementation with Webots would also be great. One excellent example of a great documentation with plenty of accessible guides and examples is the Unity ML project)

omichel, 2020-01-10:
It is certainly possible to use Webots to train robots controlled by ML via the cloud. Using extern controllers is a possibility to achieve this. However, we do not provide any ready-made toolkit or tutorial for that. But we would be happy to see some volunteers developing such a toolkit or tutorial.

closed 2020-01-10 by omichel

MentalGear, 2020-01-10:
Thank you for your answer. Indeed, I found
- "Numerical Optimization Methods"
- "Interfacing Webots to Third Party Software with TCP/IP"
in the documentation (which is one of the most well written and presented sim app guide I have seen so far! 👍) though always appreciate a hint if there's more to know.
I will try to figure out the rest, and if I succeed, might post back here with a howto. :)

DavidMansolino, 2020-01-13:
> if I succeed, might post back here with a howto. :)

This would be highly appreciated :wink: