cyberbotics/webots #1245: Webots Crash on start up 2020a Win10

MyKoo, 2020-01-08:
Hi everyone!

When I launch Webots 2020a, it shows homescreen with tutorial and then I cannot click anything, then it shuts. I don't have any reports. 

Starting Webots with safemode didn't work for me.

 - Operating System: Win 10
 - Graphics Card: NVIDIA GeForce GTX1050m

omichel, 2020-01-08:
Does Webots R2019b revision 1 work on your PC?

MyKoo, 2020-01-08:
I tried it, and unfortunetaly it doesn't. Same happened when I downloaded R2019b.

omichel, 2020-01-08:
That's odd. Are you sure your NVIDIA drivers are up-to-date?
Did you installed Webots locally (only for you) or for all users (in `C:\Program Files\Webots`)?

MyKoo, 2020-01-08:
Now, they are. It didn't help.
It's installed only for me.

omichel, 2020-01-08:
Can you try to uninstall Webots and re-install it for all users?

MyKoo, 2020-01-08:
It works, thank you very much!
If I meet further problems I will reopen this issue.

closed 2020-01-08 by MyKoo

omichel, 2020-01-08:
Can you tell me what is your local username? Does it contain any non-ASCII character, like Asian characters?

MyKoo, 2020-01-08:
ASCII characters only.

omichel, 2020-01-08:
OK, thank you for the feedback... I have no more idea to investigate the issue... We tested local installations here and they work fine... So, I have no clue on what could be the cause of the problem...

stefaniapedrazzi, 2020-01-08:
Did you have a previous Webots version installed on your system when installing Webots R2020a?

MyKoo, 2020-01-08:
I had uninstalled 2019b rev.1 before launching 2020a.
At the beggining Webots haven't worked.