cyberbotics/webots #1250: 2020a Crashes / Doesnt Save Data

MentalGear, 2020-01-09:
**Describe the Bug**
The latest stabke build frequently crashes after running and then rewinding.
Also related is that unsaved changes to the node tree disappear after running the simulation once and rewinding (not reloading). It seems like it reloads the world from last save, instead of just rewinding.

**Steps to Reproduce**
1. New Project
2. Add any node on the tree (rectangle area, light)
3. Run the sim
4. Pause & Rewind
5. Added node is deleted

**Expected behavior**
Rewind state to where it was before running the sim, without having to click save before every run.

**System**
 - mac 14.6
 - Graphics Card: Radeon Pro 570X

omichel, 2020-01-10:
It is normal that the unsaved changes to the node tree are disappearing when resetting. This is because the reset (rewind) also reverts such changes (like the reload). This is clearly [documented](https://www.cyberbotics.com/doc/guide/the-user-interface#file-menu).

Regarding the crashes you experienced. Are they totally random? Or can you describe a precise sequence (e.g., loading a specific world, modifying a specific field in a node of the scene tree manually and resetting the simulation) that yields to a systematic crash? If so, please describe this sequence, so that it can be further investigated.

MentalGear, 2020-01-10:
X for your reply, @omichel.
The docs are well done, however, this default behaviour seems rather odd, since, as a user, I don't want to constantly pay attention to remember saving the world before I run the simulation, and oftentimes, if one forgets, one can become very frustrated that the nodes one has worked on for the last few minutes are gone.
I can understand that this may be trained behaviour for experienced Webots users to always click "save" before "run", but - besides being really a big point of frustrations for new users - it's not an optimal interaction flow as it requires 2 clicks (save, run), instead of just one.
An easy fix would be to trigger a save automatically each time run is clicked.

Here's a dump of the crash:
Process:               webots [98663]
Path:                  /Applications/Webots.app/Contents/MacOS/webots
Identifier:            com.cyberbotics.webots
Version:               R2020a (R2020a)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           webots [98663]
User ID:               501

Date/Time:             2020-01-11 00:24:15.660 +0100
OS Version:            Mac OS X 10.14.6 (18G1012)
Report Version:        12
Anonymous UUID:        FE6D879B-32F0-0E37-D35E-D80BBF304C96

Sleep/Wake UUID:       C231B6DD-705A-42CE-B255-7FEEB4438B69

Time Awake Since Boot: 1900000 seconds
Time Since Wake:       36000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [98663]

VM Regions Near 0:
--> 
    __TEXT                 0000000105d5d000-0000000106201000 [ 4752K] r-x/rwx SM=COW  /Applications/Webots.app/Contents/MacOS/webots

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   com.cyberbotics.webots        	0x0000000105f4baa1 WbSolid::collectSolidChildren(WbGroup const*, bool, QVector<WbSolid*>&, QVector<WbBasicJoint*>&, QVector<WbPropeller*>&) + 449
1   com.cyberbotics.webots        	0x0000000105f4d573 WbSolid::childrenJerk() + 147
2   com.cyberbotics.webots        	0x0000000105f4d228 WbSolid::prePhysicsStep(double) + 72
3   com.cyberbotics.webots        	0x0000000105f4d3b6 WbSolid::prePhysicsStep(double) + 470
4   com.cyberbotics.webots        	0x0000000105f4d3b6 WbSolid::prePhysicsStep(double) + 470
5   com.cyberbotics.webots        	0x0000000105f4d3b6 WbSolid::prePhysicsStep(double) + 470
6   com.cyberbotics.webots        	0x0000000105f0ed43 WbRobot::prePhysicsStep(double) + 19
7   com.cyberbotics.webots        	0x0000000105f31fbc WbSimulationWorld::step() + 1516
8   com.cyberbotics.webots        	0x0000000105de8971 WbControlledWorld::step() + 1649
9   org.qt-project.QtCore         	0x0000000107618e4b QMetaObject::activate(QObject*, int, int, void**) + 1435
10  org.qt-project.QtCore         	0x0000000107620d52 QTimer::timerEvent(QTimerEvent*) + 98
11  org.qt-project.QtCore         	0x0000000107611916 QObject::event(QEvent*) + 102
12  org.qt-project.QtWidgets      	0x0000000107f5c9fd QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
13  org.qt-project.QtWidgets      	0x0000000107f5de02 QApplication::notify(QObject*, QEvent*) + 594
14  org.qt-project.QtCore         	0x00000001075e7cc4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
15  org.qt-project.QtCore         	0x000000010764139f QTimerInfoList::activateTimers() + 991
16  libqcocoa.dylib               	0x0000000112099742 0x112066000 + 210754
17  com.apple.CoreFoundation      	0x00007fff4e48fe33 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
18  com.apple.CoreFoundation      	0x00007fff4e48fdd9 __CFRunLoopDoSource0 + 108
19  com.apple.CoreFoundation      	0x00007fff4e47379b __CFRunLoopDoSources0 + 195
20  com.apple.CoreFoundation      	0x00007fff4e472d65 __CFRunLoopRun + 1189
21  com.apple.CoreFoundation      	0x00007fff4e47266e CFRunLoopRunSpecific + 455
22  com.apple.HIToolbox           	0x00007fff4d6d11ab RunCurrentEventLoopInMode + 292
23  com.apple.HIToolbox           	0x00007fff4d6d0ee5 ReceiveNextEventCommon + 603
24  com.apple.HIToolbox           	0x00007fff4d6d0c76 _BlockUntilNextEventMatchingListInModeWithFilter + 64
25  com.apple.AppKit              	0x00007fff4ba6977d _DPSNextEvent + 1135
26  com.apple.AppKit              	0x00007fff4ba6846b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
27  com.apple.AppKit              	0x00007fff4ba62588 -[NSApplication run] + 699
[... 667 lines cut ...]
MALLOC guard page                   32K        6 
MALLOC_LARGE (reserved)            288K        3         reserved VM address space (unallocated)
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
Memory Tag 242                      12K        1 
Memory Tag 249                     156K        3 
OpenGL GLSL                        512K        6 
PROTECTED_MEMORY                     4K        1 
STACK GUARD                       56.1M       26 
Stack                            140.7M       30 
VM_ALLOCATE                       49.8M      131 
VM_ALLOCATE (reserved)              36K        1         reserved VM address space (unallocated)
__DATA                            79.7M      357 
[... cut ...]
__TEXT                           401.3M      345 
__UNICODE                          564K        1 
mapped file                       94.3M       47 
shared memory                     4296K       23 
===========                     =======  ======= 
TOTAL                              1.8G     1233 
TOTAL, minus reserved VM space     1.4G     1233 

Model: iMac19,1, BootROM 220.270.93.0.0, 6 processors, Intel Core i5, 3 GHz, 8 GB, SMC 2.46f12
Graphics: kHW_AMDRadeonPro570XItem, Radeon Pro 570X, spdisplays_pcie_device, 4 GB

omichel, 2020-01-17:
The crash dump is not very helpful. Could you instead post a procedure (e.g., loading a specific world, modifying a specific field in a node of the scene tree manually and resetting the simulation) that yields to a systematic crash?

omichel, 2020-06-17:
I am closing this old PR as this crash doesn't seem to be reproducible.

closed 2020-06-17 by omichel