cyberbotics/webots #1251: Improve Name Fields Content

DavidMansolino, 2020-01-09:
As described in https://github.com/omichel/cyberbotics/pull/32#discussion_r364809321 names such as ``solid(0)``, ``solid(1)`` etc. should be avoided as they decrease the readability of the PROTO files and worlds.

Unfortunately, we currently have many of them, we should therefore improve them to put names that explain what this Solid nodes are.

omichel, 2020-01-10:
Changing manually 8000+ node names is unfortunately a considerable amount of work compared to the expected benefit (it's not really a big deal). Instead, I updated the PROTO design guidelines] in #1254 to avoid as much as possible this problem in the future.

omichel, 2020-01-10:
Addressed in #1254.

closed 2020-01-10 by omichel