cyberbotics/webots #1260: UI/API: Unit Declaration

MentalGear, 2020-01-13:
For the most part, the API/guide is well written and structured. However, sometimes the default values for properties are nowhere stated.
For example, the position sensor's resolution field simply has a float value, but not even the documentation of the position sensor specifies what unit is used. Other properties like a motor's minPosition and maxPosition also state no unit in the guide.
Ideally, each field would simply state the unit directly in app's UI.

omichel, 2020-01-15:
I am fixing these missing units in #1275. Please let me know if you see any other field for which the unit is not documented.

MentalGear, 2020-01-21:
Here's another one: https://cyberbotics.com/doc/reference/physics
Also gives no units on mass or density fields. (granted, the standard SI units for mass is kg)
Of course from a UI/UX POV, it would be best to have units displayed directly in the UI, though I can understand that this would probably be too much hassle to add.

omichel, 2020-01-21:
Addressed in #1289.

PR #1275 (2020-01-16): Added missing units to parameter descriptions
Fixes #1260.
closed 2020-01-27 by DavidMansolino