cyberbotics/webots #1261: Quality Assurance R2020a-rev1

DavidMansolino, 2020-01-13:
Since this is a very small release the list of tests is very small.

Tests:
### Installation
  * [ ] Windows
  * [ ] macOS
  * Ubuntu 16.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
  * Ubuntu 18.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
### Make a movie
  * [ ] Windows
  * [ ] macOS
  * Ubuntu 16.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
  * Ubuntu 18.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
### Visual Studio wizard
  * [ ] Windows
### Binary compilation
  * [ ] Windows
  * [ ] macOS
  * Ubuntu 16.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
  * Ubuntu 18.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
### Extern controllers
  * [ ] Windows
  * [ ] macOS
  * Ubuntu 18.04
    * [ ] Snap
### SUMO interface
  * [ ] Windows
  * [ ] macOS
  * Ubuntu 18.04
    * [ ] Snap
### Guided Tour
  * [ ] Windows
  * [ ] macOS
  * Ubuntu 16.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian
  * Ubuntu 18.04
    * [ ] Snap
    * [ ] Tarball
    * [ ] Debian

| Test | Description
|---|---|
| Installation  | Install Webots from the provided package, following the installation instruction from the Webots user guide. |
|   | Check that when pressing the simulation reset while the simulation is running you have the same behavior at the second iteration.Check this for the 10-15 first worlds of the guided tour. |
| Creation of a small project | From scratch and using only the Webots GUI (wizards (including VisualStudio on windows), scene tree, new, and so on), create a world |
| VRML import/export  | Check that the VRML import/export menu is working. |
| Make a movie  | Make a movie using the menu: Files > Make Movie ... </br>Test at least a normal screen resolution and the full screen resolution (1920x1080 Full HD). |
| Delete objects in the Scene Tree  | 1. Open a complex world </br>2. Select the last object in the Scene Tree </br>3. Hit the [Delete] button until the Scene Tree is empty (besides the 3 top nodes) </br>4. Revert and repeat the operation starting from the 4th node. </br>The test succeeds if Webots does not crash. |
| Copy a robot in the Scene Tree  | 1. Open a world that contains a robot</br>2. Select the robot</br>3. Hit the [Copy] button</br>4. Hit the [Paste After] button</br>5. At this point there should be 2 robots close to one another</br>6. Separate the 2 robots using the mouse (if needed)</br>7. Save the .wbt</br>8. Push [Revert]</br>9. Push [Run]</br></br>The test succeeds if both robots have the same appearance and behavior, etc. |
| Java controllers  | 1. Open the world: projects/languages/java/worlds/examples.wbt</br>2. Hit [Run]</br></br>The test succeeds if all 3 robots move, the camera images change and there is no error message.</br></br>3. Modify Help message (Driver.java at line 88)</br>4. Hit the [Save] button</br>5. Accept the relocation of the project proposed by Webots, in that order hit: [Yes], [OK], [Open], [OK].</br>6. Hit the [Build] button</br>7. When Webots proposes to 'Revert', hit [Yes].</br>8. Hit [Run]</br></br>The compilation test succeeds if the help message printed in the console contains the last modifications. |
| Python controllers  | 1. Open the world: projects/languages/python/worlds/examples.wbt</br>2. Hit [Run]</br></br>The test succeeds if all 3 robots move, the camera images change and there is no error message.</br></br>All the Python versions officially supported by the OS should be tested. |
| Robot window  | The generic robot window opens itself correctly, even if the simulation is stopped or at step 0 (by double clicking on the robot, or by selecting the corresponding item in the Robot menu)</br></br>Note: The e-puck, the Thymio II and the DARwIn-OP have a custom robot window.</br> |
| Devices worlds  | Run all the examples in the 'projects/samples/devices/worlds' directory and check that they all work as expected. |
| Support of UTF-8 characters  | Test that a user having UTF-8 characters in his account name can:</br>- launch Webots</br>- launch controllers</br>- Compile controllers |
| Visual Studio wizard  | Test that the VisualStudio wizard works. |
| Matlab controllers  | 1. Open the world: projects/languages/matlab/worlds/e-puck_matlab.wbt</br>2. Hit [Run]</br></br>The test succeeds if:</br>a) the e-puck robot moves</br>b) the camera and display viewports show moving images</br>c) a Matlab Window with 4 subwindows appears and all subwindows show moving graphics | 
| Mouse interactions  | Using the mouse and the keyboard, ensure that one can move the camera (translate, rotate, zoom), the objects (xz translation, lift), and apply forces/torques.</br></br>A right click in the 3D scene opens the context menu.</br></br>A double click on a robot opens the robot window. |
| e-puck - remote control  | Try to remote control an e-puck2 robot over Bluetooth |
| Binary compilation  | Test that controllers / physics plugins / controller plugins (remote control / robot windows) / libraries (qt_utils) can be cleaned and compiled smoothly. On Windows, please make sure to test by opening Webots from the icon. |
| DARwIn-OP  | to test:</br>-> the simulations do the expected thing</br>-> controllers can be modified</br>-> you can remote-control and remote-compile the real robot</br>-> same with Robotis OP2 |
| Make HTML5 animation  | Can create an animation (File / Make HTML5 animation) and play it into a web browser. |
| Streaming Server  | 1. start with the streaming server: webots --stream</br></br>2. open the web page located in resources/web/streaming_viewer/index.html and connect to Webots (using the default parameters).</br></br>3. the 3D view of Webots should display properly on the web page. It should be possible to change the viewpoint and select objects. |
| Extern controllers  | Open a simulation with a robot. Set the controller of the robot to <extern> and launch the controller from the command line. The test succeeds if the robot behaves the same as with the normal controller. |
| Offline documentation  | Check that the offline documentation is working from 'Help / offline documentation / ...' and from the context menu. |
| SUMO interface  | Open the 'projects/vehicle/highway' world:</br></br>  1.Start the simulation and check that SUMO is correctly started in GUI mode.</br>  2. Check that the version of SUMO (top of the SUMO GUI window) is the one specified in the Changelog.</br>  3. Check in Webots that other vehicles are appearing and moving.</br>  4. Check that the main vehicle is visible in SUMO as a green triangle (you may need to zoom). |
| OSM importer and SUMO exporter  | 1. Install the python dependencies as explained in https://www.cyberbotics.com/doc/automobile/openstreetmap-importer#dependencies</br></br>2. Convert the OSM file:</br></br>mkdir -p $HOME/my_project/resources/osm_files</br>mkdir -p $HOME/my_project/resources/OSM_importer</br>cp $WEBOTS_HOME/projects/vehicles/importer_demo.sh $HOME/my_project/importer_demo.sh</br>cp $WEBOTS_HOME/resources/osm_importer/config.ini $HOME/my_project/resources/OSM_importer/config.ini</br>cp $WEBOTS_HOME/projects/vehicles/resources/osm_files/village.osm $HOME/my_project/resources/osm_files/village.osm</br>cd $HOME/my_project</br>chmod u+x ./importer_demo.sh</br>./importer_demo.sh village</br></br>3. Open the generated world file and check the result is similar to the one distributed in Webots (project/vehicles/village).</br> 4. In pause mode, add the SumoInterface node and set it's 'useNetconvert' field to FALSE.</br> 5. Run the simulation and check that you can see vehicles appearing and correctly aligned on the roads. |
| Backward compatibility  | Open few worlds of the previous Webots version, and check they can be open with the rightful warnings if required. This test should particularly focus on the modifications of the tested release breaking the backward compatibility. |
| ROS controllers  | Test the distributed ROS simulations located in projects/languages/ros. The instructions are available at https://www.cyberbotics.com/doc/guide/tutorial-8-using-ros |
| Guided Tour | Check that all the worlds of the guided tour behave as expected. |

omichel, 2020-01-14:
When opening `city.wbt` (guided tour), I am getting this error:
```
ERROR: 'C:/Users/Olivier/AppData/Local/Programs/Webots/projects/objects/traffic/protos/DirectionPanel.proto': error: Template engine error: luaL_dostring error : ...ta/Local/Programs/Webots/resources/lua/liluat/liluat.lua:520: attempt to perform arithmetic on local 'line_number' (a nil value).
```

omichel, 2020-01-14:
`range_finder.wbt` attempts to save a `tiff` image and displays an error:
```
... Webots\projects\samples\devices\controllers\range_finder\range_finder.exe"
[range_finder] Error: wb_range_finder_save_image(): .tiff image not supported anymore, use .hdr instead.
```

fabienrohrer, 2020-01-14:
The tests run smoothly on macOS. I have the same issue with the range finder, but not the one with city.wbt.

DavidMansolino, 2020-01-14:
Same on Ubuntu 18.04, I am fixing the range-finder one

DavidMansolino, 2020-01-14:
I think all the issues are now fixed, I will relaunch the creation of the packages.

DavidMansolino, 2020-01-14:
Released.

closed 2020-01-14 by DavidMansolino