cyberbotics/webots #1269: sojourner.wbt doesn't look like Mars any more

omichel, 2020-01-14:
Because of recent changes in the textures / background, the ground color of sojourner.wbt is not red any more, but grey. It's a pity as it should refer to the red planet (Mars)...

DavidMansolino, 2020-01-20:
I looked for nice looking HDR background images, but it's not so easy to find a good one, until now I only found those ones: https://www.canstockphoto.fr/search.php?related_image=54674898&related_mode=artist
Ideally, we should find similar ones (but free and a more permissive license).

omichel, 2020-01-20:
Changing the color of the soil, rocks and fog can greatly improve the red planet appearance. I will make some try a post a screenshot here.

omichel, 2020-01-20:
[image]

PR #1283 (2020-01-20): Fixed red planet color
Fix #1269.
Before this PR:
[image]
After this PR:
[image]
closed 2020-01-20 by omichel