cyberbotics/webots #1279: Mass Calculations

MentalGear, 2020-01-17:
A robot or parent can be made up from many different (solid) children nodes. Each child can have a physics node that allow to define a mass. The documentation (if I remember right) states that all the parent nodes of nodes with physics must also have physics node. (a warning is printed if the mass field of these group parent/robot nodes that hod the robot's structure are empty)

I would assume that the overall parent or robot node mass would be calculated automatically based on the children nodes, yet a mass must be defined for each overlying physics node, therfore I'm wondering how masses are calculated ?

omichel, 2020-01-20:
The mass of a robot (or any other object in Webots) is computed as the sum of the masses of its all components (own Physics filed , plus children). It is documented here: https://cyberbotics.com/doc/reference/physics. If the documentation seems unclear to you, please point precisely on the paragraph that would need a better wording and possibly suggest some improvement.

MentalGear, 2020-01-21:
I guess it's more the general concept. Having to define a mass or density for a robot node (or structural nodes in general), even though it's just a structure that holds together other parts that have mass, seems unintuitive/unreal to me.

omichel, 2020-01-21:
A `Robot` node is not a just structure that holds other nodes, it has a "physical" existence as well. It is similar to any other node inheriting from the `Solid` node (including sensors and actuators). Similarly any `Solid` node may be the parent of other `Solid` children nodes. So, I see no design flaw here.

closed 2020-03-04 by omichel