cyberbotics/webots #1290: Wrong Returned Field State for PROTO Node in Extension Field

DavidMansolino, 2020-01-21:
**Describe the Bug**
When using the Supervisor API to get the state of a field of a PROTO node in an MFField of another PROTO node, the state is not updated and always return the initial value.

This is visible with this controller:
```cpp
#include <webots/Supervisor.hpp>

using namespace webots;

int main(int argc, char **argv) {
  Supervisor *supervisor = new Supervisor();

  int timeStep = (int)supervisor->getBasicTimeStep();

  while (supervisor->step(timeStep) != -1) {
    Field *rotationField = supervisor->getSelf()->getField("rotation");
    const double *rotation = rotationField->getSFRotation();

    std::cout<<"Current rotation ! "<<rotation[0]<<" "<<rotation[1]<<" "<<rotation[2]<<" "<<rotation[3]<<" "<<std::endl<<std::endl;
  };

  delete supervisor;
  return 0;
}
```
And this world:
```
#VRML_SIM R2020a utf8
WorldInfo {
}
Viewpoint {
  orientation -0.6103433939870436 0.7734836625371138 0.17089167681472558 0.6570328003602236
  position 1.7142384103450126 2.350849111004101 2.7804630483690542
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
}
E-puck {
  translation 0 0.95 0
  controller "my_controller"
  supervisor TRUE
  turretSlot [
    E-puck {
      translation 0 0.06 0
      controller "my_controller"
      supervisor TRUE
    }
  ]
}
```
The rotation field of the second e-puck always returns the initial value even if manually changed in the scene-tree while the simulation is running.
It works fine for the first e-puck and if either the first or the second e-puck is converted to base nodes.

stefaniapedrazzi, 2020-08-06:
I cannot reproduce this issue on the current master branch (R2020b).

If the fields are not manually changed from the scene tree and given that the internal robot translation and rotation values are relative to the parent robot it is correct that they do not change (they match the ones in the scene tree).

If I manually change the translation or rotation values from the scene tree, the values printed by the controller are updated in the next step.
If the simulation is in pause mode and I change the values, then two steps are needed to see the new values in the console.

DavidMansolino, 2020-08-06:
I just tried and I confirm it is now working fine, this was probably fixed in the meantime 😃

closed 2020-08-06 by DavidMansolino