cyberbotics/webots #1296: Broken docs Links with Tab

DavidMansolino, 2020-01-22:
**Describe the Bug**
When entering a link to the doc with the tab-os or tab-language set, the page is not found and the index is displayed:
https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-os=macos&tab-language=python

DavidMansolino, 2020-01-27:
It seems there is an issue on the PHP side when parsing the url.
It seems that the `tabs` are not correctly parsed and are part of the `page`:
```
Setup: {"book":"guide","page":"running-extern-robot-controllers?tab-os=macos&amp;tab-language=python","tab":"","tabs":[],"anchor":"","branch":"hotifx-doc","repository":"cyberbotics","tag":"","url":"https://raw.githubusercontent.com/cyberbotics/webots/"}
```

Here is the problematic parsing:
https://github.com/cyberbotics/webots/blob/0d7f111588ab876434c39f995b8c1e5736088144/docs/doc.php#L28-L74

stefaniapedrazzi, 2020-01-27:
maybe we should replace
```
"/&(tab-[^=?&]+)=([^?&#]+)/"
```
by 
```
/[&?](tab-[^=?&]+)=([^?&#]+)/"
```

DavidMansolino, 2020-01-27:
Yes, that's probably the cause. Not sure what is the procedure to test changes on this file?

stefaniapedrazzi, 2020-01-27:
I don't remember how much this functionality can be tested locally.
For sure it is possible to test part of it.
I can look at this if you want.

DavidMansolino, 2020-01-27:
If you have an environment to test set-up why not.

stefaniapedrazzi, 2020-01-27:
Ok.
Yes, I can easily test this using the local robotbenchmark setup.

DavidMansolino, 2020-01-27:
Perfect, thanks a lot.

PR #1303 (2020-01-29): Documentation: fix extracting tabs if version is not specified
Fix #1296.

The code to extract queries if the version is not specified has some issues:
* `$version` is not defined 
* regular expression should be adjusted

Moreover the code to parse the query is quite complex.
Using PHP built-in functions to parse the URL and the query string it is possible to simplify it.

URL to check the functionality (after uploading the new version on the server):
https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-os=macos&tab-language=python
closed 2020-01-29 by stefaniapedrazzi