cyberbotics/webots #1310: Add Supervisor Internal Field Functions

DavidMansolino, 2020-01-29:
**Describe the Bug**
Currently, a supervisor can't parse himself if it is a PROTO.
It would be very convenient to allow a supervisor to have access to all its fields and node in read-only mode.

For this, two new functions should be added:
```
WbNodeRef wb_supervisor_node_get_from_def_internal(const char *def);
WbFieldRef wb_supervisor_node_get_field_internal(WbNodeRef node, const char *field_name);
```

Their usage will be exactly the same as for the non-'_internal' ones.

omichel, 2020-01-29:
I would suggest a different naming:
```
WbNodeRef wb_supervisor_node_get_from_proto_def(const char *def);
WbFieldRef wb_supervisor_node_get_proto_field(WbNodeRef node, const char *field_name);
```
This clearly refers to PROTO and is more understandable to me.
What do you think?

DavidMansolino, 2020-01-29:
Better indeed!

closed 2020-02-11 by DavidMansolino