cyberbotics/webots #1316: Webots wont import proto with connector

Acwok, 2020-02-03:
**Describe the Bug**
Webots wont import proto with connector as child.

**Steps to Reproduce**
1. Create a proto based on a solid with a connector as child
2. try to import it in Webots with the add new button
3. the proto doesn't appear in the list

Tested on webots 2019b revision1, webots 2020a, webots 2020a revision1

**Proto example:**

```
#VRML_SIM R2019b utf8
# license: Apache License 2.0
# license url: http://www.apache.org/licenses/LICENSE-2.0

PROTO Test [
    field  SFVec3f     translation  0 0 0
    field  SFRotation  rotation     0 1 0 0
    field  SFString    name         "test"
]
{
    Solid {
        translation IS translation
        rotation IS rotation
        children [
            Connector {       
            }
        ]
    }
}
```

DavidMansolino, 2020-02-04:
Thank you for this bug report, I can indeed reproduce this. The PROTO can only be inserted in a Robot node whereas it should be possible to insert it as a root node.

DavidMansolino, 2020-02-04:
This problem was fixed in #1317, the fix will be available in the next version of Webots (and the nightly builds).

closed 2020-02-04 by DavidMansolino