cyberbotics/webots #1336: MATLAB R2019b is not detected by Webots on Windows 10

wbadry, 2020-02-06:
**Describe the Bug**
MATLAB R2019b is not detected by Webots on Windows 10

**Steps to Reproduce**
1. run **cmd** or **PowerShell**
2. type **matlab**, it should open up MATLAB.
[image]

3. Open **C:\Program Files\Webots\projects\languages\matlab\worlds\e-puck_matlab.wbt**
[image]

4. See error
**WARNING: To run Matlab controllers, you need to install Matlab 64-bit and ensure it is available from the DOS CMD.EXE console.**

**Expected behavior**
MATLAB controller script is running normally

**System**
 - Operating System: Windows 10
 - Graphics Card:  NVIDIA GeForce 1050 Ti

DavidMansolino, 2020-02-06:
Maybe link with #681

wbadry, 2020-02-07:
Thank you @DavidMansolino for your response. Could you at least try it using a trial version? Thanks

omichel, 2020-02-10:
We are currently in discussion with the Mathworks on this topic. We will keep you informed here.

wbadry, 2020-02-12:
Thanks Oliver.

sxbug, 2020-02-17:
This question is different from  #681
 #681 problem is that as soon as I start the simulation, matlab will be black and stuck。
I have found some faults in launcher.m and  allincludes.h , then I modified : the webots made a mistake in the controller directory of MATLAB (it is not Webots\lib\matlab  ,it is Webots\lib\controller\matlab). After I modified it, there were other errors.
when I change controller to <extra>,and run launcher.m in Matlab 2017b ,the error msg displayed:

>> launcher
The MATLAB "MinGW-w64 C/C++ Compiler" addon is not installed, please install it from: https://fr.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler
Using prototype file: C:\Users\xsun\AppData\Local\Temp\protofile_matlab_2017b_webots_R2020a_rev2.m
error use loadlibrary
load library  "Controller"  error
The specified module could not be found.

error launcher (line 126)
    loadlibrary(libname,protofile,'alias','libController');

Reason:
Wrong use of loaddefinedlibrary
The specified module could not be found.

I thought It's probably the problem with loadlibrary
Can I help you?  Thanks

DavidMansolino, 2020-02-19:
We fixed many issues with Matlab in #1378, we have successfully tested it with the latest version of Matlab (R2019b update 4).

wbadry, 2020-03-09:
Hello @DavidMansolino 
I downloaded the nightly build 6/3/2020. Still having same issue on **MATLAB 2019b**.
[image]
I rechecked that MATLAB can run using **matlab** command from CMD window:
[image]

omichel, 2020-03-10:
I checked this and I can run the nightly build (6/3/2020) of Webots R2020a-rev2 together with MATLAB R2019b without any problem. The only difference I have with your setup is that my path only include `C:\Program Files\MATLAB\R2019b\bin` and not `C:\Program Files\MATLAB\R2019b\runtime\win64`. I tried to add it, but MATLAB still works fine with Webots...
Did you install Webots for all users or just for one user?
Did you set the `WEBOTS_HOME` environment variable globally to point to `C:\Program Files\Webots` or the local user installation folder?

wbadry, 2020-03-10:
I will walk you through what I did so maybe a hidden bug is there in the release.

1. I installed the nightly build (6/3/2020) .
2. I followed the [tutorials ](https://cyberbotics.com/doc/guide/tutorial-1-your-first-simulation-in-webots) using Pyhton with no issues and MATLAB. I named MATLAB controller **EPuckGoForward* * and copied the MATLAB script.
3. When nothing is happening, I went to launcher.m to test it from MATLAB.
4. Before running it, I checked path and environment variables needed. Note that MATLAB is under "Polyspace" not "MATLAB"
[image]
[image]
[image]
5. I started to run launcher with breakpoints to get the variables retrieved from the environment.
[image]

WEBOTS_HOME path is correct
[image]
 
6. I ran into the second section getting the version of WEBOTS and changing MATLAB working directory to webots matlab controller library. Since I call the script from MATLAB, test_mode is set to true.
[image]

I think the problem is in the **protofile** name as I don't think backslash is allowed
[image]
[image]

I don'y know how it runs on your machine. For some reason, MATLAB splits the name after month and that's why it shows that this function is unknown.

sxbug, 2020-03-11:
Maybe you not set  Environment variables :WEBOTS_CONTROLLER_NAME?
WEBOTS_CONTROLLER_NAME=matlab

omichel, 2020-03-11:
This [patch](#1444), to be included in the next nightly build, should fix the protofile issue. But normally, this code shouldn't be executed if you define properly `WEBOTS_CONTROLLER_NAME` to run your controller as an extern controller.
From your description, I understand you cannot run your own MATLAB controller, but can you run the MATLAB example provided in `$WEBOTS_HOME/projects/languages/matlab/worlds/e-puck_matlab.wbt`?

omichel, 2020-03-11:
I tested with MATLAB R2017b and for some reason, it doesn't work. MATLAB R2017b hangs at start-up. However, MATLAB R2019b works nicely.

wbadry, 2020-03-11:
> This [patch](#1444), to be included in the next nightly build, should fix the protofile issue. But normally, this code shouldn't be executed if you define properly `WEBOTS_CONTROLLER_NAME` to run [...]

I tried nao_matlab example by setting MATLAB as external controller. All variables were set as in [document](https://www.cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-language=matlab) .
[image]

The launcher script is still broken until it is fixed. When I tried to run the MATLAB controller from MATLAB, it reported that libraries are not found.
[image]

Is there any additional setting I should make? Libraries are not found as I believe launcher is supposed to load it in temp directory (or perhaps not)

It would be great if there is a video setting for MATLAB as it might br a missing configuration that was skipped in documentation.

wbadry, 2020-03-11:
> 
> 
> I tested with MATLAB R2017b and for some reason, it doesn't work. MATLAB R2017b hangs at start-up. However, MATLAB R2019b works nicely.

Hi @omichel 
Could it be due to **folder naming change** from **MATLAB** to **Ployspace** on R2019b?

wbadry, 2020-03-11:
> 
> 
> Maybe you not set Environment variables :WEBOTS_CONTROLLER_NAME?
> WEBOTS_CONTROLLER_NAME=matlab

Even in external mode, the protofile is still broken and maybe that's why the controller script is unable to load controller library files

omichel, 2020-03-11:
I don't think that the folder naming change to **Ployspace** is a problem.
However, in your environment variable settings, you should set `WEBOTS_CONTROLLER_NAME` to `nao_matlab` and `WEBOTS_PROJECT` to `C:\Program Files\Webots\projects\robots\softbank\nao` to run the nao matlab demo.
And be sure that Webots is indeed installed for all users in `C:\Program Files\Webots` and not for a specific user (in `C:\Users\MyUsername\AppData\Local\Programs\Webots`).

omichel, 2020-03-11:
By the way, you should probably set the environment variables from MATLAB as shown [here](https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-language=matlab#environment-variables).

wbadry, 2020-03-11:
> I don't think that the folder naming change to **Ployspace** is a problem.
> However, in your environment variable settings, you should set `WEBOTS_CONTROLLER_NAME` to `nao_matlab` and [...]

Thanks @omichel 
I was using the example in the languages folder for e-puck as a test. I committed the changes as instructed to check nao example
[image]

The library path needed to be added (MATLAB detected it). However, after I run it, I got the 

```
following error:
Error using calllib
Library was not found

Error in wb_robot_get_device (line 6)
result = calllib('libController', 'wb_robot_get_device', name);

Error in nao_matlab (line 18)
camera = wb_robot_get_device('CameraTop');
```

wbadry, 2020-03-11:
Was the version written correctly or should add nightly build?

omichel, 2020-03-11:
The version is correct (it should include nightly build).

wbadry, 2020-03-11:
Does `libController `reside somewhere or it is created in runtime by loadlibrary ?

omichel, 2020-03-11:
`libController` resides in `$WEBOTS_HOME\lib\controller` and is actually named `Controller.dll`. It should be in your `PATH` environment variable as explained [here](https://cyberbotics.com/doc/guide/running-extern-robot-controllers#environment-variables), so that MATLAB can load it.

wbadry, 2020-03-11:
I guess the header file is missing to use loadlibrary and then calling the calllib?

omichel, 2020-03-11:
I don't think the header file is needed.
As I said, it works nicely on my machine and also on the machine of @DavidMansolino.
I am not sure what is the problem on your machine.
I see that `C:\cygwin64\bin` is in your `PATH`. This may cause some conflicts. Can you try to remove it and see if it works? Or maybe another folder in your `PATH` is causing a problem (loading some wrong DLL)?

sxbug, 2020-03-12:
Thanks @omichel
I want to know whether only the matlab 2017B is not valid？

omichel, 2020-03-12:
I tested only MATLAB 2017b and 2019b and it turns out that 2017b doesn't work for me while 2019b works well. I didn't test other versions.

sxbug, 2020-03-12:
Thanks @omichel
I would wait for you fix this problem.

omichel, 2020-03-12:
Sorry, but we do not plan to fix it. I would recommend you to upgrade to a more recent version of MATLAB.

sxbug, 2020-03-12:
OK. Thanks

wbadry, 2020-03-12:
Hi,
Can we at least get a confirmation that MATLAB has no issues from anyone else other than developers? Sometimes, the developer machine has installed files that don't exist on production machine. I would appreciate this.

sxbug, 2020-03-14:
So disappointed, I installed matlab '9.7.0.119022 (r2019b)' . But MATLAB R2019b is not detected by Webots on Windows 10 too. There was an error, just like wbadry's.

WARNING: To run Matlab controllers, you need to install Matlab 64-bit and ensure it is available from the DOS CMD.EXE console.

webots ：2020b nightly build 13/3/2020

SYSTEM：
Operating System: Windows 10 enterprise Chinese Version
Graphics Card: NVIDIA GeForce 750 Ti

PATH: D:\Program Files\Polyspace\R2019b\bin;  
          D:\Program Files\Polyspace\R2019b\polyspace\bin;  
          D:\Program Files\Polyspace\R2019b\runtime\win64

WEBOTS_HOME=D:\Program Files\Webots
WEBOTS_VERSION=2020b
WEBOTS_CONTROLLER_NAME=MATLAB

mex -setup
      MEX 配置为使用 'MinGW64 Compiler (C)' 以进行 C 语言编译。
      要选择不同的语言，请从以下选项中选择一种命令:
       mex -setup C++ 
       mex -setup FORTRAN

How can I do?   Thanks

sxbug, 2020-03-14:
only set e-puck controller=<extern>. 
in CMD windows input: matlab -nosplash -nodesktop -minimize -r launcher
then can start matlab simulate.

wbadry, 2020-03-14:
@sxbug This is fantastic work!
I tried this on nao and it works although the launcher script displayed "C++ compiler" is not installed while it is installed and verified using  `mex -setup` in MATLAB

[image]
[image]
[image]

Actually, in an easier way, you can run launcher directly from MATLAB. The issue is I was trying to run the **nao_matlab script directly from MATLAB** which fails. Using launcher as mentioned is a great solution. Still to be noted that Webots can not directly run MATLAB scripts like python and C files.

To make this thread useful. Let's summarize **how to run MATLAB scripted Webots on Windows**. 

**Example folder:** 
C:\Program Files\Webots\projects\robots\softbank\nao
or
WEBOTS_HOME \projects\robots\softbank\nao

1. Download and install MinGW-w64 C/C++ compiler from **MATLAB Addons**

[image]

2.  Look inside the **controllers folder**. Your MATLAB controller file must match the folder name. In this example, the script name and its containing folder is named **nao_matlab**

[image]
[image]

3. Now both **worlds** and **controllers** are ready. Time to verify **environment variables** to match your project settings

3.1 path settings

[image]

3.2 variables settings
pay attention to **WEBOTS_CONTROLLER_NAME** and **WEBOTS_VERSION**

[image]

5. Restart your computer

6. Set nao controller to  `<extern>`

[image]

7. Open MATLAB and type `help launcher`

[image]

If MATLAB doesn't recognize launcher, you may need to add `C:\Program Files\Webots\lib\controller\matlab` or ` WEBOTS_HOME \Webots\lib\controller\matlab` to MATLAB search path.

8. run Webots project

[image]

9. from MATLAB command window, type **launcher** and hit **enter**

[image]

You should see a message showing the **protofile script** generated.

And NAO robot should start walking while MATLAB plots  feet pressure, accelerometer, camera feedback, and edge detection 

[image]

wbadry, 2020-03-14:
Thanks, @omichel for being patient with me. Hopefully, we can identify the issue and solve it somehow in the next releases. I would be happy to participate if needed. Thanks to you and @DavidMansolino for being active and responsive to all issues even though it is free.

sxbug, 2020-03-14:
If 'mex - setup' checked  is OK , it doesn't matter if the warning msg 'mingw-w64 C / C + + compiler addon is not installed' is displayed  during simulation. Matlab Simulation is available.

omichel, 2020-03-16:
@wbadry: thank you for this detailed report. Can you now launch the MATLAB examples from Webots? To do so, you should remove the following environment variables: `WEBOTS_CONTROLLER_NAME`, `WEBOTS_PROJECT` and `WEBOTS_VERSION` and open the original `nao_matlab.wbt`, e.g., without the controller set to `<extern>`.

wbadry, 2020-03-16:
> @wbadry: thank you for this detailed report. Can you now launch the MATLAB examples from Webots? To do so, you should remove the following environment variables: `WEBOTS_CONTROLLER_NAME`, [...]

Unfortunately, it is still not working. 
@sxbug were you able to run MATLAB controller directly from Webots?

closed 2020-03-16 by omichel

sxbug, 2020-03-16:
@wbadry Matlab 2019 can't be started directly from webots, or it can only be started in extra mode and from CMD （input : matlab .... -r launcher）. But now I can start matlab 2017b directly, just comment out lines 28-32 in launcher. m.

sxbug, 2020-03-16:
It's very convenient to use, but there is still a library that hasn't been loaded successfully. What's the problem?

[matlab]  (dynamic library)
[matlab] Error: remote control initialisation failed
[matlab] Error: Cannot load the "C:/Program Files/Webots/projects/robots/gctronic/e-puck/plugins/remote_controls/e-puck_bluetooth/e-puck_bluetooth.dll" remote control library.
[matlab] Running Matlab sample Webots controller.

sxbug, 2020-03-16:
I can not upload any PICs

omichel, 2020-03-16:
You can ignore this e-puck_bluetooth library. It is used only for remote-control of an e-puck robot. It shows up probably because of a DLL conflict in your PATH. I don't have it on my system.

sxbug, 2020-03-16:
The DLL file is in this directory ‘D:/Program Files/Webots/projects/robots/gctronic/e-puck/plugins/remote_controls/e-puck_bluetooth/’. I checked it. That is OK
Thanks
@omichel