cyberbotics/webots #1344: error using python

Eli7S, 2020-02-10:
INFO: my_controller: Starting controller: python.exe -u "my_controller.py" 
[my_controller] Traceback (most recent call last):
[my_controller]   File "my_controller.py", line 7, in <module>
[my_controller]     from controller import *
[my_controller]   File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 15, in <module>
[my_controller]     import _controller
[my_controller] ImportError: DLL load failed: The specified module could not be found.
WARNING: 'my_controller' controller exited with status: 1.

how to fix it? I'am beginner with webots

omichel, 2020-02-11:
Did you install the 64 bit version of Python 3.7 from [python.org](https://www.python.org/downloads/release/python-376/)? The direct link to the download is [here](https://www.python.org/ftp/python/3.7.6/python-3.7.6-amd64.exe).
What do you get when you type `python -V -V` in a DOS console?
Can you check if you can execute the python demo? It is located in [C:\Program Files\Webots\projects\languages\python\worlds\example.wbt](https://github.com/cyberbotics/webots/blob/master/projects/languages/python/worlds/example.wbt). If you can run it, then, you should be able to run your own python simulation.

Eli7S, 2020-02-11:
python -V -V results: 
Python 3.7.6 (tags/v3.7.6:43364a7ae0, Dec 19 2019, 01:54:44) [MSC v.1916 64 bit (AMD64)].
I tried to execute the python demo, but also I faced the same problem(same error);

omichel, 2020-02-11:
Did you installed Webots locally (for the current user) or globally (for all users, e.g., in `C:\Program Files\Webots`)?

Eli7S, 2020-02-11:
globally for all users!!

omichel, 2020-02-11:
Which version of Webots are you using? R2020a revision 1? Or a nightly build?

omichel, 2020-02-11:
I just tested on Windows 10 with Webots R2020a revision 1 and Python 3.7.6 and everything works smoothly for me. Maybe you could try it with Webots R2020a revision 2 (nightly build from yesterday) and Python 3.8.1? That configuration works for me as well.

Eli7S, 2020-02-11:
Thank you. It seems like I have some problems with python. I reinstalled Python 3.7.6, now It's working perfectly.

closed 2020-02-11 by omichel