cyberbotics/webots #1345: Build webots from source

rachel-dev, 2020-02-10:
Hello everyone,

I try to build webots from source. But when I do ```make```, I got error ```recipe for target '/usr/local/webots/dependencies/webots-qt-5.13.1-linux64-release.tar.bz2' failed```.

I try to build it from source because I try to write my controller with ROS 1 in Python 2.7. But when I tried to run controller I got ```ImportError: /usr/local/webots/lib/controller/python27/_controller.so: undefined symbol: PyClass_Type```

Does any have any ideas?

Does anyone have ideas?

`

stefaniapedrazzi, 2020-02-10:
If you cloned the Webots repository in "/usr/local" folder it is likely that you have issues with write permissions.
Did you check if you can create and modify the Webots files as normal user? 

If not I would recommend you to clone the Webots repo in a folder where you have writing rights.

rachel-dev, 2020-02-10:
Before I kept webots which I installed from debian package. Source code of webots is in my ```home``` folder.
 Now I use ```sudo apt remove webots``` and then did ```make ``` again. I got ```Makefile:54: /usr/local/webots/resources/Makefile.os.include: No such file or directory```

rachel-dev, 2020-02-10:
I found problem. I need to set up ```WEBOTS_HOME``` first

closed 2020-02-10 by rachel-dev