cyberbotics/webots #1347: error: when running external controller with ros1

rachel-dev, 2020-02-10:
Hello everyone,

I try to run ros as extern controller. I used example universal_robots_ros.py. I set up environment in Pycharm. When I run this file, I got
 ```universal_robots_ros.py", line 27, in <module>
    rospy.init_node('UR5e', disable_signals=True)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 336, in init_node
    raise rospy.exceptions.ROSInitException("Failed to initialize time. Please check logs for additional details")
rospy.exceptions.ROSInitException: Failed to initialize time. Please check logs for additional details```

Anyony has ideas?

DavidMansolino, 2020-02-11:
Before trying to run it from Pycharm, have you tried running it from the terminal?

rachel-dev, 2020-02-11:
Hi David, I tried running it from the terminal. It works now! Thank you very much.

closed 2020-02-11 by rachel-dev