cyberbotics/webots #1350: Controlling robotiq 2 finger gripper: joint didn't move other parts of gripper

ipa-rwu, 2020-02-11:
Hello everyone,

the main goal is to control robotiq 2 finger gripper from moveit.

I am trying to use robotiq 2 finger 85 gripper in webots. I used ```urdf2webots``` transfer gripper urdf which is in this  [repository](https://github.com/StanleyInnovation/robotiq_85_gripper/tree/master/robotiq_85_description/urdf) to ".proto" file which you can find in this [link](https://github.com/ipa-rwu/ur5_2f_webots/blob/master/protos/Robotiq85Gripper.proto). 

Now I can connect it with moveit. I sent a goal from rviz as you can see this figure. It moves perfect in rviz.
[image]
[image]

But there is only one joint moves in webots. It supposes "gripper_finger1_joint" move other parts. But it didn't. As you can see this picture. 
[image]
Is it model probleam? I have no ideas to config proto files.

Anyone has ideas?

DavidMansolino, 2020-02-11:
There is indeed probably an issue with your model.
Would you be interested in adding your model to Webots?
If so, please open a PR to add your PROTO to this folder https://github.com/cyberbotics/webots/tree/master/projects/devices/robotiq/protos
Once the PR is open I will help you fixing your model.

ipa-rwu, 2020-02-11:
Thank you very much. I already did.

DavidMansolino, 2020-02-11:
Perfect, I will have a look and let you know directly on the PR.

DavidMansolino, 2020-07-14:
Linked to #1365

ipa-rwu, 2020-08-03:
Thank you for releasing a new version. I tried the model for robotiq 2 finger gripper which was created for 2020a but is not working for 2020b yet. You can find in this [link](https://github.com/cyberbotics/webots/tree/ipa-rwu-patch-1/projects/devices/robotiq/protos)

DavidMansolino, 2020-08-04:
What is not working exactly?
Which difference can you experience between R2020a-rev1 and R2020b?

P.S. feel free to add your model in https://github.com/cyberbotics/community-projects

ipa-rwu, 2020-08-04:
If I use it in R2020b, I got these errors:
```
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':26:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_0' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':46:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_1' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':66:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_2' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':86:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_3' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':106:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_4' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':126:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_5' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':168:24: error: Skipped unknown 'robotiq_arg2f_85_outer_knuckle_0' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':192:28: error: Skipped unknown 'robotiq_arg2f_85_outer_finger_0' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':234:34: error: Skipped unknown 'robotiq_arg2f_85_inner_finger_0' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':359:24: error: Skipped unknown 'robotiq_arg2f_85_inner_knuckle_0' node or PROTO.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':474:38: error: Skipped unknown 'Box' field in Shape node.
ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':474:42: error: Expected field name or '}', found '{'.
```

DavidMansolino, 2020-08-04:
It seems the `robotiq_arg2f_85_base_link_XXX` PROTO files are not found, do they exist?

ipa-rwu, 2020-08-06:
> It seems the `robotiq_arg2f_85_base_link_XXX` PROTO files are not found, do they exist?

I don't have these files. I just use the one you helped me to improve it before as you can find here. https://github.com/cyberbotics/webots/tree/ipa-rwu-patch-1/projects/devices/robotiq/protos

DavidMansolino, 2020-08-06:
The PROTO raising these error is the ``Robotiq2f.proto``one, I don't see it here: https://github.com/cyberbotics/webots/tree/ipa-rwu-patch-1/projects/devices/robotiq/protos

DavidMansolino, 2020-08-13:
> If I use it in R2020b, I got these errors:
>
> ```
> ERROR: 'ur5_2f_webots/protos/Robotiq2f.proto':26:18: error: Skipped unknown 'robotiq_arg2f_85_base_link_0' node or PROTO.
> ERROR: [...]

@ipa-rwu this is now fixed in https://github.com/cyberbotics/community-projects/pull/10, there were some issues with missing DEF-USE.

omichel, 2020-11-30:
I suppose we can close this issue. If not, please feel free to re-open it or better, open a new one as this one is pretty long and it's not clear what was the exact problem.

closed 2020-11-30 by omichel