cyberbotics/webots #1355: From the point of (inverse kinematics) How can i know real torque

rrkblee3, 2020-02-12:
(1) I made "2 link manipulator" because I have to experiment inverse kinematics test.
so i could get simulate torque was calculated by input angle. 
but it is not accurate. SO i wander if 'webots' can give me some method to get accurate torque, 
when i set my  motor parts by input angle

(2) When i set my motor with input angle. It was not coincide with real calculated angle 
     like this picture. how can i set my motor with input as it is.

 
[image]

DavidMansolino, 2020-02-12:
Have you tried increasing the `maxVelocity` field of your motor?

rrkblee3, 2020-02-13:
> d of your motor?

Yes but It was not coincide  real calculated angle and input angle still....

DavidMansolino, 2020-02-13:
Would it be possible for you to share your simulation?

rrkblee3, 2020-02-17:
> Would it be possible for you to share your simulation?
sure, thanks Mr.D  
[2LP.zip](https://github.com/cyberbotics/webots/files/4211419/2LP.zip)

DavidMansolino, 2020-02-17:
Thank you, we will have a look at it and let you know here.

DavidMansolino, 2020-02-17:
I checked your simulation, and the behavior you are observing looks normal for me, since you are constantly changing the target position, it is normal that the actual position is always 'late'.
If I set the target position to a fixed value, then after a transitional period (which is characteristic of any PID controller) the exact target value is reached.
If you want to get an exact and immediate match between your target and actual position, you have to do a kinematic simulation, but this will be way less realistic.

closed 2021-03-01 by stefaniapedrazzi