cyberbotics/webots #1359: Implement RotationalMotorLink & LinearMotorLink

omichel, 2020-02-13:
To address the problem mentioned in #1351, we should implement two new nodes: `RotationalMotorLink` and `LinearMotorLink` that would refer to an exiting motor node and would be set in the `device` list of a joint node. Such nodes would be somehow similar to the `SolidReference` node, but with extra parameters defining the mechanical link between the original motor and the secondary joint.

This would simplify, for example, the design of grippers where the two (or three) fingers are driven by the same motor. Similarly, the articulated fingers of robot hand could be driven by a single motor.

```
RotationalMotorLink {
  SFString motorName ""  # name of the mechanically linked motor
  SFFloat multiplier 1   # gear/pulley ratio (may be negative to reverse motion)
  SFFloat gain 0.95      # energy gain (should be < 1 for purely mechanical system)
}
```

Other fields may be added later on such as `minPosition` or `maxPosition`.

DavidMansolino, 2020-02-13:
I am wondering if we should limit this to controlled joints. It might be usefull to extend this to free joints (URDF is supporting this for example).

omichel, 2020-02-14:
Replaced with #1365 which supports free joints.

closed 2020-02-14 by omichel