cyberbotics/webots #1361: Webot ROS connection crashes when supervisor is on the robot and you call the service to get the translation of the node through the supervisor service

hkdeman, 2020-02-13:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2020-02-13:
Can you please edit your post to add the description in the post instead of the title and if possible provide some code and stack.

hkdeman, 2020-02-13:
> Can you please edit your post to add the description in the post instead of the title and if possible provide some code and stack.

Apologies! Changed!

DavidMansolino, 2020-02-14:
Can you check what is the value of the `node` argument of this service that you are using?

closed 2020-02-14 by DavidMansolino