cyberbotics/webots #1368: Wireframe color won't reset to white after some collision

Acwok, 2020-02-14:
**Describe the Bug**
Sometimes when a solid enter in collision with another one, its wireframe stay red until the user unselect the solid and select it again.

**Steps to Reproduce**
1. in a zero gravity world with two solids
2. select a solid
3. apply a force to make them enter in collision. the selected solid becomes red.
4. sometimes the selected solid stay red until the user click away

**Expected behavior**
The Wireframe color turn back to white after collision

**Screenshots**
https://youtu.be/W2Se0ipM7nQ

DavidMansolino, 2020-02-14:
Would it be possible to display the contact points in the video too?

Acwok, 2020-02-14:
The option was enabled while shooting the video

DavidMansolino, 2020-02-14:
Sorry I didn't look carefully enough.
This is indeed definitely a bug.

DavidMansolino, 2020-02-17:
I can reproduce a similar issue with the `extended_solids` world, by setting the gravity to 0, and applying a force on one of the objects so that it collides another.
Can you confirm that if you un-select and re-select the object the color is then correct?

Acwok, 2020-02-17:
Yes same issue with the `extended_solid` world. I confirm that if I un-select and re-select the object, the color is then correct.

DavidMansolino, 2020-02-17:
Thank you for the confirmation, I have been able to isolate the problem a bit more, this happens only when a collision lasts only one time step.

DavidMansolino, 2020-02-18:
Fixed in #1375

closed 2020-02-18 by DavidMansolino