cyberbotics/webots #1369: Can not find Qt5QmlModels.dll, when execute webots

sxbug, 2020-02-15:
**Describe the Bug**
Operating System:Win 10 64Bit Enterprise Chinese Version
Webots 2020a Rev2
Graphics Card: AMD R9 M730X

The installation is completed normally, an error is prompted “Code execution cannot continue because the file  Qt5QmlModels.dll cannot be found, reinstall  may fix this problem”  when Webots startup, click OK and the Webots Prog exits .

sxbug, 2020-02-15:
I downloaded it from Internet .
Qt5QmlModels.dll 
Version:5.14.0.0; product name:Qt5  ; Size: 409KB
But when I run Webots, an error msg displayed:The application failed to start properly(0xc0000007b). Please click OK to close the application.
Thanks

omichel, 2020-02-17:
Can you try with this version of Qt5QmlModels.dll and let me know if that resolves the problem?
[Qt5QmlModels.zip](https://github.com/cyberbotics/webots/files/4212368/Qt5QmlModels.zip)

sxbug, 2020-02-17:
Yes. 
This problem has been solved. Successfully used python3.8.
Thanks very much!!!!!

closed 2020-02-17 by sxbug

omichel, 2020-02-17:
OK, so that's apparently a new dependency of Qt5 on Windows. I will fix it for the next nightly builds.

omichel, 2020-02-17:
Fixed in #1373.