cyberbotics/webots #1372: Crash when double click on Camera Display

sudhirkelappan, 2020-02-17:
**Describe the Bug**
Webots crashes when I double click on Camera Display

**Steps to Reproduce**
1. webots --mode=pause /usr/local/webots/projects/vehicles/worlds/city.wbt
2. In the world view double click on the camera display overlay to open it in a separate window.
4. Webots coredumps.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Linux Ubuntu 16.04
 - Graphics Card: NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2

**Additional context**
Happens on Speedometer display overlay as well.

sudhirkelappan, 2020-02-17:
[backtrace_from_coredump.txt](https://github.com/cyberbotics/webots/files/4213669/backtrace_from_coredump.txt)

DavidMansolino, 2020-02-17:
Probably linked to #302.

stefaniapedrazzi, 2020-09-21:
Should be fixed in #2119.

closed 2020-09-21 by stefaniapedrazzi