cyberbotics/webots #1384: Emitter/Receiver communication between nodes not working after controller and simulation reset

tsampazk, 2020-02-18:
**Description**
Main Issue:
I'm using a supervisor node to reset a robot controller and the simulation. The supervisor node and the robot node communicate with receivers/emitters. They work ok up until i reset the robot controller. In the dumbed down example below the robot is sending a counter to the supervisor,
who resets the simulation and the robot controller when it receives the value 10.

After resetting the robot controller and it is restarted and properly broadcasting, it seems that the supervisor receiver stops receiving the messages (the queue length stays at 0). After several steps (12 for me in the example below) the supervisor receiver starts to pick up the old messages (after reset) in order, thus they get badly desynchronized. The send method in the robot emitter properly returns 1.

What i found:
The issue seems to happen when the simulation is also reset via the supervisor.simulationReset()
method is called (it is irrelevant if its called before or after resetting the robot controller. If removed, the expected behavior happens.

Side issue:
Moreover, after the robot controller reset, the supervisor receives an extra value that it probably shouldn't. According to the docs, the sent messages are received instantly and the emitter/receiver queues are cleared by the simulationReset() method.

Based on the main + side issues my guess is that the receiver/emitter queues are messed up after the simulationReset() method.

Note:
In the very first step, the supervisor doesn't receive a value, probably depending on the order the step functions of the two controllers are called.

**Steps to Reproduce**
World .wbt file:
```
#VRML_SIM R2020a utf8
WorldInfo {
}
Viewpoint {
  orientation -0.6931032800836722 0.6931032800836722 0.1980295085953349 0.75
  position 1.2 1.6 2.3
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
}
DEF ROBOT Robot {
  translation 0.16 0.08 0
  children [
    Receiver {
    }
    Emitter {
    }
  ]
  controller "robot_controller"
}
DEF SUPERVISOR Robot {
  translation -0.22 0.08 0.1
  children [
    Receiver {
    }
    Emitter {
    }
  ]
  name "supervisor"
  controller "supervisor_controller"
  supervisor TRUE
}
```
robot_controller.py
```
from controller import Robot

class RobotEmitter:
    def __init__(self):
        self.robot = Robot()

        self.timestep = int(self.robot.getBasicTimeStep())

        self.emitter = self.robot.getEmitter("emitter")

        self.data = 0

    def get_timestep(self):
        return self.timestep

    def run(self):
        while self.robot.step(self.timestep) != -1:
            self.data += 1
            self.handle_emitter()

    def handle_emitter(self):
        string_message = str(self.data)
        string_message = string_message.encode("utf-8")
        print("robot handle emitter string message to send:", string_message)
        if self.emitter.send(string_message):
            print("message sent")
        else:
            print("emitter queue full")

robot_controller = RobotEmitter()
robot_controller.run()
```
supervisor_controller.py
```
from controller import Supervisor

class SupervisorReceiver:
    def __init__(self):
        self.supervisor = Supervisor()
        self.timestep = int(self.supervisor.getBasicTimeStep())

        self.robot = self.supervisor.getFromDef("ROBOT")
        
        self.receiver = self.supervisor.getReceiver("receiver")
        self.receiver.enable(self.timestep)
        self.receivedData = ""

    def get_timestep(self):
        return self.timestep

    def run(self):
        while self.supervisor.step(self.timestep) != -1:
            self.handle_receiver()

            if self.receivedData == "10":
                print("RESET")
                self.supervisor.simulationReset()  # This causes the issue, if you comment it out it works as expected
                self.robot.restartController()

    def handle_receiver(self):
        if self.receiver.getQueueLength() > 0:
            self.receivedData = self.receiver.getData().decode("utf-8")
            print("supervisor handle receiver data:", self.receivedData)
            self.receiver.nextPacket()
        else:
            print("supervisor receiver q is empty")

supervisor = SupervisorReceiver()
supervisor.run()
```
**Expected behavior**
The supervisor receives the values the robot is emitting when they are emitted after resetting
the robot controller and the simulation

**System**
 - Windows 10
 - Webots version R2020a rev 1

DavidMansolino, 2020-02-18:
Thank you for the precise bug report, I can indeed reproduce the issue.

DavidMansolino, 2020-02-18:
It seems the problem is coming from this condition which is always true after the reset:
https://github.com/cyberbotics/webots/blob/18751886a0f25f0b247c24e3ba688c265c3dcdfc/src/webots/nodes/WbReceiver.cpp#L347

DavidMansolino, 2020-02-18:
This is because the supervisor controller is not restarted.
A workaround (which is not ideal) would be to restart the supervisor controller too:
supervisor_controller.py
```Python
class SupervisorReceiver:
    def __init__(self):
        self.supervisor = Supervisor()
        self.timestep = int(self.supervisor.getBasicTimeStep())

        self.robot = self.supervisor.getFromDef("ROBOT")
        
        self.receiver = self.supervisor.getReceiver("receiver")
        self.receiver.enable(self.timestep)
        self.receivedData = ""

    def get_timestep(self):
        return self.timestep

    def run(self):
        while self.supervisor.step(self.timestep) != -1:
            self.handle_receiver()

            if self.receivedData == "10":
                print("RESET")
                self.supervisor.simulationReset()  # This causes the issue, if you comment it out it works as expected
                self.robot.restartController()
                self.supervisor.getSelf().restartController()

    def handle_receiver(self):
        if self.receiver.getQueueLength() > 0:
            while self.receiver.getQueueLength() > 0:
                self.receivedData = self.receiver.getData().decode("utf-8")
                print("supervisor handle receiver data:", self.receivedData)
                self.receiver.nextPacket()
        else:
            print("supervisor receiver q is empty")

supervisor = SupervisorReceiver()
supervisor.run()
```

tsampazk, 2020-02-18:
Well, for our specific use case this is not possible unfortunately. We are running a reinforcement learning agent on the supervisor and we need the supervisor to control the simulation and reset it without it being reset too. 

We already have a different implementation of resetting, which is more cumbersome and less general but works for now. Thank you for taking the time to find a workaround though.

DavidMansolino, 2020-02-19:
This is fixed in #1385, the patch will be available in the nightly build from tomorrow.

tsampazk, 2020-02-20:
@DavidMansolino I installed the 19/02/20 nightly version and the main issue is fixed.
However, the side issue (of an extra value sent after resetting the robot controller) still persists. 

I wanted to ask if this is intended or it's something i am misunderstanding in the way emitters/receivers work etc. Thank you.

DavidMansolino, 2020-02-20:
@tsampazk when running your example I get this:
```
...
[supervisor_controller] supervisor handle receiver data: 8
[robot_controller] robot handle emitter string message to send: b'10'
[robot_controller] message sent
[supervisor_controller] supervisor handle receiver data: 9
[supervisor_controller] supervisor handle receiver data: 10
[supervisor_controller] RESET
INFO: robot_controller: Starting controller: python3.7 -u "robot_controller.py" 
[supervisor_controller] supervisor handle receiver data: 11
[supervisor_controller] supervisor receiver q is empty
[robot_controller] robot handle emitter string message to send: b'1'
[robot_controller] message sent
[supervisor_controller] supervisor handle receiver data: 1
[robot_controller] robot handle emitter string message to send: b'2'
[robot_controller] message sent
...
```

Do you get the same?
This perfectly makes sense for me. I assume that the `supervisor handle receiver data: 11` is perturbing you, but this can be explained:

### step N-1:
Robot sends 10.
Supervisor receives 9.

### step N:
Robot sends 11.
Supervisor receives 10.
Supervisor reset world.
Supervisor restart Robot.

### setp N + 1:
Supervisor receives 11 because it was sent in the pipe before the Robot was restarted.
Robot sends 1.

Does this makes sense for you.

tsampazk, 2020-02-20:
I get exactly the same. Yes ```supervisor handle receiver data: 11``` seemed irregular to me.

Step N-1 seems ok and is backed up by the printed messages. Step N however doesn't correspond to the printed messages. Shouldn't the robot print out that it sent "11"? Probably between supervisor receiving "9" and "10", same as the previous prints? I can see that the robot step happens first, then the supervisor step (as far as i can tell it's alphabetical?). If something is sent from the robot, then the supervisor doesn't immediately get it, because those robot handle_emitter is run after the robot.step and the emitter needs to be stepped for the packet to actually be sent. The supervisor gets the message on its next step and that's what the [documentation](https://www.cyberbotics.com/doc/reference/emitter#wb_emitter_send) calls "instant":

> Note that independently from the baudRate, the Emitter node will need at least one basic time step to send the packet but the Receiver node will receive it immediately.

I can see that happening if i flip the call order of robot.step and handle_emitter.

```Python
def run(self):
    while True:
        print("handling emitter")
        self.handle_emitter()
        print("stepping")
        if self.robot.step(self.timestep) == -1:
            break
        self.data += 1
        print("just stepped")
```

Moreover, the documentation reads that simulationReset empties the emitter/receiver queues, so even if robot sent "11" in step N with ```self.emitter.send("11")```, intuitively, it should be lost during simulation reset. I guess after "11" is sent and until robot controller is reset, "11" isn't inside any queue (it's in the pipe?). But the  [documentation](https://www.cyberbotics.com/doc/reference/emitter#wb_emitter_send) reads:

>The wb_emitter_send function **adds to the emitter's queue** a packet of size bytes located at the address indicated by data. The enqueued data packets will then be sent* to potential receivers (and removed from the emitter's queue)

\* i guess this means not when emitter.send() but at the next robot.step as the earlier quote suggests

So from what i can tell things happen in this order:
### Step N - 1
- robot.step runs ("9" arrives at supervisor)
- robot sends "10"
- supervisor receives 9
### Step N
- robot steps ("10" actually arrives at supervisor receiver)
- robot sends "11" without printing and "11" should still be in emitter queue according to the documentation, needs one more step
- supervisor receives "10"
- supervisor resets simulation, clearing queues
- supervisor calls robot.resetController
### Step N + 1
- robot.step runs ("11" actually arrives at supervisor receiver, queue not cleared or "11" was somewhere else)
- robot controller actually resets
- supervisor receives "11"
### Step N + 2
- robot.step runs
- robot sends "1"
- supervisor receives nothing

Note that both supervisor and robot nodes have synchronization set to TRUE, so that means we don't get any weird concurrency issues, just the controllers gettings stepped in a certain (alphabetical?) order, which also changes (as expected) after the first reset.

A workaround for my example is to step the supervisor after reset, to "dump" the rest of the queue which is somewhat weird, because this snippet is already nested inside a "while step":
``` Python
self.supervisor.simulationReset()
self.robot.restartController()

while self.supervisor.step(self.timestep) != -1:
    if self.receiver.getQueueLength() > 0:
        self.receiver.nextPacket()
    else:
        break
```

DavidMansolino, 2020-02-21:
> Step N however doesn't correspond to the printed messages. Shouldn't the robot print out that it sent "11"? Probably between supervisor receiving "9" and "10", same as the previous prints?

You are completely right, this is wrong and `[robot_controller] robot handle emitter string message to send: b'11'` should be displayed. I just fixed this in #1394, the problem was that the robot controller process stdout was not flushed before restarting it, the stdout was therefore lost.

I will now investigate why the package '11' is not actually removed from the queue (or, as you said, it's also highly probable that the packet is somewhere else at this point in time).

>  I can see that the robot step happens first, then the supervisor step (as far as i can tell it's alphabetical?)

The order is the scene-tree order, since `ROBOT` is higher in the scene-tree than `SUPERVISOR` it will run first.

By the way, it seems you have a very good understanding of how Webots works internally!

tsampazk, 2020-02-21:
Thank you very much! I have used physics engines, etc., in the past and have an idea of how they do stuff.

DavidMansolino, 2020-02-21:
I have checked what is happening, here are the steps:

### N - 1
  - robot and supervisor steps.
  - robot sends '10' to Webots.
  - supervisor reads '9'.
  - Webots transmits '10' to supervisor.

### N
  - robot and supervisor steps.
  - robot sends '11' to Webots.
  - supervisor reads '9'.
  - Webots transmits '10' to supervisor.
  - supervisor sends robot restart request to Webots.
  - supervisor sends reset request to Webots.
  - Webots reset the simulation (and therefore empty the receiver/emitter queues for remaining packets on the Webots side).
  - Webots restart the robot controller

### N + 1
  - robot and supervisor steps.
  - robot init (nothing is sent).
  - supervisor reads '11' which was received before the reset.

### N + 2
  - robot and supervisor steps.
  - robot sends '1' to Webots.
  - supervisor reads nothing.
  - Webots transmits '1' to supervisor.

### N + 3
  - robot and supervisor steps.
  - robot sends '2' to Webots.
  - supervisor reads '1'.
  - Webots transmits '2' to supervisor.

If we would like to avoid this '11' message to be received this would imply that at reset Webots should send a request to the supervisor process to empty the queue of received messages already transmitted. However, I am not sure this is the desired behavior, as messages already received should probably remain if the controller is not restarted.
Does this make sense for you?

tsampazk, 2020-02-21:
So between step N and step N + 1, "11"  is received in receiver queue, but queue doesn't actually empty from the simulation reset method?

I was working off the [user guide](https://www.cyberbotics.com/doc/guide/the-user-interface#file-menu), that reads that at simulation reset the queues are emptied on nodes, i assume regardless whether the node's controller is reset or not, plus the [documentation ](https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_simulation_reset)reading:
>The wb_supervisor_simulation_reset function sends a request to the simulator process, asking it to reset the simulation immediately. The reset process is explained in detail in the User Guide, **the only difference** is that the supervisor and robot controllers are not restarted, if needed, they have to be restarted with the wb_supervisor_node_restart_controller function.

DavidMansolino, 2020-02-24:
There was indeed a small issue in the documentation, the reset is not applied immediately but only once the step is over. This is now fixed in the documentation: https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_simulation_reset

tsampazk, 2020-02-24:
Ok then, i think the issue is resolved completely now. Thank you David!

DavidMansolino, 2020-02-25:
You're welcome.

tsampazk, 2021-01-25:
Hello again! 

We ran into an issue related to this one when trying to reset the world via a supervisor getting messages from a receiver. In the specific use case, a robot sends a position sensor values via an emitter to a supervisor node which gets the value via a receiver. When the value surpasses a threshold, the supervisor resets the world.

The issue lies in the '11' value being referenced in previous examples in the thread. The robot  sends the first value that surpasses the threshold, let's say the threshold is 0.10 and the robot sends 0.105. When this value is received a reset is triggered, but the robot sends the next value, let's say 0.11 in that step. This value is not cleared by the reset and is received after it, causing an additional reset. Moreover, another extra value is sent during the second reset causing a third reset, even though the world was reset fully during the first reset. We couldn't find what seems to cause this second additional reset. 

A more high level explanation can be found in [this issue](https://github.com/aidudezzz/deepbots/issues/51).

Quoting from [earlier](https://github.com/cyberbotics/webots/issues/1384#issuecomment-589596497):
> If we would like to avoid this '11' message to be received this would imply that at reset Webots should send a request to the supervisor process to empty the queue of received messages already [...]

As mentioned here this behaviour might be the desired one for Webots. However, there does not seem to be another straightforward way to clear these additional messages from the queues, other than restarting both controllers, which actually defeats the purpose of resetting the world only. Below i am including the world and controller files including two fixes in the supervisor controller. Sadly, in our use-case it is not possible to restart the supervisor controller, thus we are left with the second solution which is highly unreliable.

If this behaviour is actually the correct one from Webot's side, is there any other possible solution apart from the two in the example below? 

During the critical reset step, the additional value does not seem to be inside the receiver queue yet, so it must be somewhere else, where we don't have access. Is there a chance to have access to some method that forcibly clears all messages in queues or sent messages residing in other places?

Lastly, this behaviour does not seem to be clearly explained in the documentation [here](https://cyberbotics.com/doc/reference/supervisor#wb_supervisor_simulation_reset) and especially in the [user guide](https://www.cyberbotics.com/doc/guide/the-user-interface#file-menu) where it is stated that emitter/receiver queues are cleared by the reset. Messages not in queues, but sent during the step are not mentioned at all. Maybe indeed, the world is reset first clearing queues and the message is received by the supervisor afterwards.

Thanks in advance!

.wbt file:
```#VRML_SIM R2021a utf8
WorldInfo {
  coordinateSystem "NUE"
}
Viewpoint {
  orientation -0.9793101212699826 0.18812350611173317 0.07457367381590291 0.6769308310076335
  position 0.333675195497526 1.896325798980918 1.485738950901334
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
}
DEF SUPERVISOR Robot {
  children [
    Receiver {
    }
  ]
  name "supervisor"
  controller "supervisor_controller"
  supervisor TRUE
}
DEF ROBOT Robot {
  translation 0 0.24 0
  children [
    Emitter {
    }
    DEF POLE HingeJoint {
      jointParameters HingeJointParameters {
        position -4.812689688184746e-16
        axis 0 0 1
        anchor 0 0.03000000000047226 0
        minStop -1.3
        maxStop 1.3
      }
      device [
        DEF POLE_POS_SENSOR PositionSensor {
          name "polePosSensor"
        }
      ]
      endPoint Solid {
        translation 0.0886013000000228 0.5142019999999938 -8.072739999999879e-16
        rotation -1.1580691675597052e-15 1.3600652487584567e-15 1 1.3943692965781191
        children [
          DEF POLE Shape {
            appearance PBRAppearance {
              baseColor 1 0.1568627450980392 0.1568627450980392
[... 99 lines cut ...]
                # Solution #2
                # The "flushing" loop fixes the issue for this 
                #  simple use-case without needing restarts. 
                #  It captures the two extra messages. This is 
                #  unreliable, non-general, and breaks down 
                #  when more messages are being exchanged.
                # for _ in range(2):
                    # if self.supervisor.step(self.timestep) == -1:
                        # exit()
                    # print("flushing")
                    # self.handle_receiver()
                # End solution #2

    def handle_receiver(self):
        if self.receiver.getQueueLength() > 0:
            self.receivedData = float(self.receiver.getData().decode("utf-8"))
            print("supervisor handle receiver data:", self.receivedData)
            self.receiver.nextPacket()
        else:
            self.receivedData = 0.0
            print("supervisor receiver q is empty")

supervisor = SupervisorReceiver()
supervisor.run()
```

stefaniapedrazzi, 2021-01-25:
I will check the issue and see what are the options to fix it.
I had a quick look and it seems reasonable that also the messages on the way to be transmitted should be cleared during the reset.

stefaniapedrazzi, 2021-02-04:
I inspected the message transmission in more details.
The third reset message received is clearly a bug of the reset functionality and I could easily fix it.

About the second reset message (the "11" message), it is currently consistent with the reset procedure where the simulation reset is applied at the very end of the step after sending the step updates to the controller, i.e. after forwarding the "11" or second reset message to the controller.

I agree this could not be the expected behavior and I'm now analyzing the implications of resetting the simulation before sending the step updates to the controller (but it seems this breaks other simulations).

closed 2021-02-10 by stefaniapedrazzi

stefaniapedrazzi, 2021-02-10:
@tsampazk I tested your example and the issue should now be fixed by #2720.

tsampazk, 2021-02-10:
> @tsampazk I tested your example and the issue should now be fixed by #2720.

Thank you very much! Is it available in the 9/2 nightly? I tried it out in `webots-R2021b_setup.exe
` and nothing seems changed. Did i download the wrong version? :sweat_smile:

stefaniapedrazzi, 2021-02-10:
I merged the fix this morning in the R2021a revision1 and I will now port the changes to the R2021b version so that tomorrow it will be available in both the 10/2 nightlies.

PR #2720 (2021-02-10): Improve synchronization of Supervisor simulation reset
Fixes #1384: fix reset behavior.

Apply the reset at the very end of the step on Webots side, but before sending the new simulation status to the controllers.
Before the reset was applied after sending the new simulation status to the controllers, i.e. values on the controller side were referring to the status prior to the reset for one additional step after the reset was requested.