cyberbotics/webots #1387: Windows installer: loading errors when updating Webots version

stefaniapedrazzi, 2020-02-19:
**Describe the Bug**
If the user modifies manually the files in the installation folder (e.g. `C:/Program Files/Webots`), the uninstaller doesn't remove completely all the files.

If some files cannot be deleted, the error should be written the uninstaller logs in red.
This is probably not sufficient and if possible it would be better to prompt the error in a message box window or make it more visible so that users cannot miss it.

This is already the case for example when the uninstaller tries to remove a directory that doesn't exists.

**Users report**
We regularly have users reports about error after updating the Webots version.

For example a user reported that he modified the robots/sphero project by editing the world file and adding extra texture.
After updating to a newer version of Webots, there were errors loading the BB-8 PROTO node in a custom world.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Windows 10

omichel, 2020-02-19:
It is indeed possible to remove all the files, but do we really want to do that? See this [discussion](https://stackoverflow.com/questions/643547/how-to-configure-inno-setup-to-uninstall-everything).

stefaniapedrazzi, 2020-02-19:
No, I don't think that removing all the files and deleting the users modifications without notifying him is better. And modifying the installation files is not a good practice that Webots should really support.

But I think that we should improve the notifications to the user, like forcing him to take some specific actions, instead of simply logging the errors because users don't read them and then reports issues and ask for help.

PR #1793 (2020-06-17): Windows uninstaller warns and delete the whole modified installation folder
Fixes #1387.

On Windows, when uninstalling Webots, the uninstaller will remove all the files that were installed by the installer. Then, if some files remain in the Webots installation folder (because they were added or modified by the user), the uninstaller will warn the user about it and ask confirmation to delete the whole installation folder.
closed 2020-06-17 by omichel