cyberbotics/webots #1396: Support for external meshes

omichel, 2020-02-21:
In order to facilitate import from different 3D formats and to ease the maintenance of PROTO and world files, we should support external meshes with various format: DAE, STL, OBJ, etc.
The `Mesh` node could be defined as follow:

```
Mesh {
  MFString url [ ]       # URL of a 3D file in DAE, STL or OBJ format (similar to ImageTexture node)
  SFFloat creaseAngle 0  # used if no normals are provided (similar to IndexedFaceSet node)
  SFBool normalPerVertex TRUE # used if normals are provided (similar to IndexedFaceSet node)
}
```

DavidMansolino, 2020-03-17:
Partially implemented in #1419.
A second PR should implement the `creaseAngle` and `normalPerVertex` fields (the Mesh node was already implemented in a way that it will be easy to add them).

omichel, 2021-01-28:
I don't believe it's needed to add the `creaseAngle` and `normalPerVertex` to the `Mesh` node as most 3D files format (like OBJ) should include pre-computed normals. Using pre-computed normals makes that Webots loads meshes twice as fast as with a crease angle computation, which we want to encourage.

closed 2021-01-28 by omichel