cyberbotics/webots #1399: " DLL load failed" when using pycharm

Weiyi-Zhang258, 2020-02-21:
Hello, when I execute python scripts controllers in webots, it works well.
But when I follow the instructions about how to use pycharm on webots, it does not work.
I'm using win10.

[image]

[image]

omichel, 2020-02-21:
Which version of Webots are you using? Did you try with a nightly build?

omichel, 2020-02-21:
I believe I found the problem: you should add an extra PATH to your environment variables:
`PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller\;G:\Program Files\Webots\msys64\mingw64\bin`

Can you try that and confirm it works?

Weiyi-Zhang258, 2020-02-22:
> I believe I found the problem: you should add an extra PATH to your environment variables:
> `PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller\;G:\Program Files\Webots\msys64\mingw64\bin`
> 
> Can you try that and confirm it works?

Yes!!! I have tried your solution and it works very well!!! Thank you very much for your detailed help!

Weiyi-Zhang258, 2020-02-22:
> I believe I found the problem: you should add an extra PATH to your environment variables:
> `PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller\;G:\Program Files\Webots\msys64\mingw64\bin`
> 
> Can you try that and confirm it works?

Hi, there is another problem. After I succeeded in running the example code, I tried to control a robotis-op2 robot in pycharm. And I'm sure to add the extra path you mentioned, but it seems some more PATH needs to be added. Can you tell me what should I add?
[image]
[image]
[image]

Weiyi-Zhang258, 2020-02-22:
> > I believe I found the problem: you should add an extra PATH to your environment variables:
> > `PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller\;G:\Program Files\Webots\msys64\mingw64\bin`
> > Can you try that and confirm it works?
> 
> Hi, there is another problem. After I succeeded in running the example code, I tried to control a robotis-op2 robot in pycharm. And I'm sure to add the extra path you mentioned, but it seems some more PATH needs to be added. Can you tell me what should I add?
> 
> The same error occurs at here:
> from managers import RobotisOp2GaitManager, RobotisOp2MotionManager
> 
> Traceback (most recent call last):
> File "D:/controllers/op2_controller2/op2_controller2.py", line 16, in
> from managers import RobotisOp2GaitManager, RobotisOp2MotionManager
> File "G:\Program Files\Webots\projects\robots\robotis\darwin-op\libraries\python37\managers.py", line 15, in
> import _managers
> ImportError: DLL load failed: 找不到指定的模块。
> 
> Here are my settings:
> PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller;G:\Program Files\Webots\msys64\mingw64\bin

I have found it:
G:\Program Files\Webots\projects\robots\robotis\darwin-op\libraries\robotis-op2

closed 2020-02-22 by Weiyi-Zhang258

LIU-yx97, 2020-10-22:
> I believe I found the problem: you should add an extra PATH to your environment variables:
> `PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller\;G:\Program Files\Webots\msys64\mingw64\bin`
> 
> Can you try that and confirm it works?

LIU-yx97, 2020-10-22:
> Yes!!! I have tried your solution and it works very well!!! Thank you very much for your detailed help!

I also met the same problem, and modified according to your way, but the problem still exists.
This is my Environment variables:
PYTHONUNBUFFERED=1;
PATH=E:\Program Files1\webots\Webots\lib\controller\;E:\Program Files1\webots\Webots\msys64\mingw64\bin\;E:\Program Files1\webots\Webots\msys64\mingw64\bin\cpp\;;
PYTHONPATH=E:\Program Files1\webots\Webots\lib\controller\python37

PR #1402 (2020-02-24): Fixed PyCharm documentation
Fixes #1399.