cyberbotics/webots #1412: project.json file

omichel, 2020-02-26:
In order to prepare an online service that would run simulations from git repositories, it would be useful to introduce a configuration file for Webots projects. Basically a git repository would correspond to a Webots project folder. At the root level (in addition to the usual `controllers`, `protos`, `worlds`, etc. folders) we could add an optional `project.json` file that may contain the following optional information:

```json
{
  "title":"My wonderful robot simulation",
  "summary":"In this simulation, a robot is trying to catch a ball.",
  "author":"Olivier Michel",
  "version":"1.0",
  "platforms":[
    {
      "name":"ubuntu",
      "version":"18.04",
      "minVersion":"16.04"
    }
  ],
  "dependencies": [
    {
      "name":"webots",
      "version":"R2020a-rev2",
      "minVersion":"R2020a"
    }
  ],
}
```

If the `platform` field is omitted, the project is supposed to run on any platform where it is possible to install the `dependencies`. Otherwise, the project needs to be run on one of the specified platforms.

`platforms` and `dependencies` specifications include mandatory `name` and `version` fields and an optional `minVersion` field. The `version` field refers to the preferred version needed to run the project. If `minVersion` is specified, the project is supposed to work on all versions between `minVersion` and `version` inclusive. An optional `maxVersion` could be introduced later on.

More information should be provided in the standard README.md file, including citations.

DavidMansolino, 2020-02-27:
About the dependencies, it might be interesting to define the type too (e.g. `pip`, `snap` etc.).

stefaniapedrazzi, 2020-02-27:
What about how to specify which world to run (at least first)?
The `worlds` folder could contain many files.
Would it be an additional parameter to pass along with the GitHub repo?

omichel, 2020-02-27:
Thank you both for the comments. Here is an updated version that takes into account your input:
```json
{
  "default":"worlds/my_wonderful_simulation.wbt",
  "platforms":[
    {
      "name":"ubuntu",
      "version":"18.04",
      "minVersion":"16.04"
    }
  ],
  "dependencies": [
    {
      "name":"webots",
      "type":["deb","snap"],
      "version":"R2020a-rev2",
      "minVersion":"R2020a"
    }
  ]
}
```
I believe the title, description and author(s) should be retrieved from the default world in `WorldInfo.title` and `WorldInfo.info` to avoid duplication and encourage users to set these fields appropriately. Also, I have some doubts about the usefulness of the `version` field, as in most cases version management is better handled by a git tag or other versioning mechanisms. Users are still free to put a version number in `WorldInfo.info` or in `README.md` if needed.

omichel, 2022-06-13:
This is not relevant any more as we are now using dockers in online simulations.