cyberbotics/webots #1416: getVelocity Clearification

MentalGear, 2020-02-27:
**Documentation page(s) concerned**
https://www.cyberbotics.com/doc/reference/motor?tab-language=python#wb_motor_get_velocity

**What is the problem with the current documentation?**
As a user, I would expect the getVelocity method to return the current velocity of the motor, not the targeted one. This can lead to confusion when looking to get the current speed.

Futhermore, the docs describe:

The wb_motor_get_target_position function allows the user to get the target position. This value matches with the argument given to the last wb_motor_set_position function call.

> The specified velocity can be retrieved using the wb_motor_get_velocity function. The wb_motor_get_max_velocity function returns the limit specified in the maxVelocity field. Note that if the velocity is not explicitly set using the wb_motor_set_velocity function, then the wb_motor_get_velocity and wb_motor_get_max_velocity functions return the same value.

The documentation gives no clear indication that this is actually the set velocity (or max) that is returned, not the current one (only the "specified", which is easy to skip).

**Describe how you would fix it**
Introduce getTargetVelocity (in line with getTargetPosition) to clear up what is returned. Link getVelocity with a note to getTargetVelocity (for backwards usage). Use getTargetVelocity as the official name in the documentation.

omichel, 2020-02-28:
Thank you for this suggestion which is very relevant. I believe we have a similar problem with the acceleration. Here is my suggestion to fix it more generally and consistently:

Deprecate the following functions:

`wb_motor_set_position` &rarr; replace with `wb_motor_set_target_position`.
`wb_motor_set_velocity` &rarr; replace with `wb_motor_set_target_velocity`.
`wb_motor_set_acceleration` &arr; replace with `wb_motor_set_target_acceleration`.
`wb_motor_get_velocity` &rarr; replace with `wb_motor_get_target_velocity` .
`wb_motor_get_acceleration` &arr; replace with `wb_motor_get_target_acceleration`.

MentalGear, 2020-02-29:
Yes, agreed with the function renaming.

Since most robots use servomotors now, it might also be worth adding a "servoMotor" node / motor subclass which encapsulates convenience functions that actually get the current acceleration/velocity etc. ( getCurrentVelocity(), ... ).
This would also make it easier to get up and running for both novices and experts.

Furthermore, a native, low-level implementation of those basic functions (which are calculated in most implementations on every step) might bring a not insignificant performance boost.

omichel, 2020-03-02:
A `Servo` node is provided in [projects/devices/generic/protos/Servo.proto](https://github.com/cyberbotics/webots/blob/master/projects/devices/generic/protos/Servo.proto). Then, it's fairly easy to implement a servo API layer on the top of the motor and position sensor APIs.

MentalGear, 2020-03-02:
I assume that's a yes for a servo base node?

omichel, 2020-03-02:
No. We won't implement a `Servo` base node as it would be redundant with the `Motor` and `PositionSensor` nodes. Rather the Webots proto system allows us to create a `Servo` proto that combines both the `Motor` and the `PositionSensor` into a single node and provide users with a simple interface. The performance overhead is very little and concerns only loading the proto file. I believe it's the way to go for users looking for a simple interface for creating servo motors.

MentalGear, 2020-03-02:
Alright, base nodes are only primitives. But a standard proto item within the node browser then ?

omichel, 2020-04-15:
Yes, it is already available in the node browser:
`PROTO nodes (Webots Projects)` &rarr; `devices` &rarr; `generic` &rarr; `Servo (Transform)`.

closed 2020-04-15 by omichel