cyberbotics/webots #1417: Is `Track.textureAnimation` really needed?

omichel, 2020-02-28:
I am wondering whether we could save the user from setting the `Track.textureAnimation` field. Can't we compute the value of this field automatically? It seems that the motion of the texture should always be in the same way given the orientation of the track.

stefaniapedrazzi, 2020-02-28:
I'm not sure it can really be computed automatically in such a way that works in all the cases.
Also because a point to take into consideration is that, at least for me as user of Webots, it is much simpler to adapt the `Track.textureAnimation` scale factor rather than modifying the texture mapping that could be very complex.
But I will make some tests to check if there is a way to compute it automatically so that it works for all the possible cases.

Also note, that the `Track.textureAnimation` field also lets the user disable the texture animation, so even if the animation factor is automatically computed it should be replaced by a SFBool field.

omichel, 2020-02-28:
> Also because a point to take into consideration is that, at least for me as user of Webots, it is much simpler to adapt the Track.textureAnimation scale factor rather than modifying the texture mapping that could be very complex.

But, that would be simply to move the `textureAnimation` values to the `TextureTransform.scale` or so, isn't it?

DavidMansolino, 2020-03-02:
With @michou214 we just found that there is a bug with this field because the speed of the texture depends on the world time step.
This is because of this line of the `WbTrack::prePhysicsStep` function:
https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbTrack.cpp#L617
The duration of the step is not taken into account for the translate of the texture.
Instead it should be:
```diff
diff --git a/src/webots/nodes/WbTrack.cpp b/src/webots/nodes/WbTrack.cpp
index 468139e..6f82b2e 100644
--- a/src/webots/nodes/WbTrack.cpp
+++ b/src/webots/nodes/WbTrack.cpp
@@ -614,7 +614,7 @@ void WbTrack::prePhysicsStep(double ms) {
 
   // texture animation
   if (mTextureTransform) {
-    mTextureTransform->translate(mSurfaceVelocity * mTextureAnimationField->value());
+    mTextureTransform->translate(0.001 * ms * mSurfaceVelocity * mTextureAnimationField->value());
     mTextureTransform->modifyWrenMaterial(mShape->wrenMaterial());
   }
```

With this patch, setting the `textureAnimation` to 1.0 seems to work (i.e. match with the physics behavior) in all cases.

remi-cyberbotics, 2020-03-18:
:eyes:

stefaniapedrazzi, 2020-03-18:
I still don't think that the textureAnimation field can be completely removed.
This field helps user to avoid having to change the texture mapping of the mesh and this won't change even after fixing the issue in the prePhysicsStep function.

omichel, 2020-03-18:
Isn't it better to set this mapping in a `TextureTransform` node instead?

stefaniapedrazzi, 2020-03-18:
Are you sure you can achieve the exact same behavior applying an texture transform?
Because I'm not sure it is the same.
If I need to move my texture in the y direction instead of the x direction it doesn't seem the same as rotating the texture (for a non symmetric texture).

DavidMansolino, 2020-03-18:
Actually I just found that the patch I posted in my previous post is good because it fixes the dependency on the time step, but we can't set the texture animation to 1.0 in all cases, it still depends on the 'length' of the track.
So maybe what we can do is to apply this patch and update the worlds/proto but keep the field?

omichel, 2020-03-18:
But if it depends on the length of the track, can't we compute this length and use it to set the animation parameter automatically?

DavidMansolino, 2020-03-18:
Yes, for a simple box case it is easy (for example in the ConveyorBelt PROTO this is already taken into account: https://github.com/cyberbotics/webots/blob/77ccaf095c3384f82fb4c91e81dccde1e51259be/projects/objects/factory/conveyors/protos/ConveyorBelt.proto#L96), but this might quickly get tricky to compute if the box is not axis aligned and even worse for more complex geometries.
And it doesn't solve the problem @stefaniapedrazzi is mentioning (which I have to admit is still not 100% clear to me).

omichel, 2020-03-18:
Neither for me... OK, let's fix this dependency on the time step first and we will see later if we can get rid of this textureAnimation parameter.

stefaniapedrazzi, 2020-03-19:
The `textureAnimation` parameter doesn't only specify the "speed" of the animation, but also the direction, i.e. if the texture is translated in the u, v or a mix of two direction.
The issue I'm trying to describe is that in complex meshes I don't see how you can automatically compute the orientation of the mesh texture w.r.t to the motor move direction in order to automatically get the translation unit vector need to animate the texture.

Also note that we would in any case still need a field to enable/disable the texture animation.

remi-cyberbotics, 2020-04-09:
Should we close this issue ? It seems it hasn't been done automatically...

omichel, 2020-04-09:
No, your PR fixed a bug with the Track node, but did not answered the original question posted here. I will assign myself to this issue as @stefaniapedrazzi already answered, but I still want to investigate a little bit.