cyberbotics/webots #1418: Custom roscpp Controller doesn't work when not using kinetic

PymZoR, 2020-02-28:
**Describe the Bug**
Webots override `LD_LIBRARY_PATH` when running a controller. The ROS shared libraries shipped with webots in `projects/default/libraries/ros` are compiled against the kinetic release. When using another release, weird behaviors appears, such as `ros::Time::now()` either yielding random values, or blocking silently indefinitly (this was hard to track down ! thanks @DavidMansolino )

**Steps to Reproduce**
Run the following controller using e.g Melodic
```Cpp
#include <ros/ros.h>
#include <iostream>
#include <memory>
#include <webots/Robot.hpp>

using namespace webots;

int main(int argc, char** argv) {
    std::shared_ptr<Robot> robot(new Robot());

    ros::init(argc, argv, "test_node");
    ros::console::set_logger_level(ROSCONSOLE_DEFAULT_NAME, ros::console::levels::Debug);
    ros::NodeHandle nh;

    while (robot->step(1000) != -1) {
        std::cout << ros::Time::now() << std::endl;
        ros::spinOnce();
    }

    return 0;
}
```

**Expected behavior**
Multiple solutions are possible :
- Try to find the user's ROS distribution based on e.g `ROS_DISTRO=$(ls -1 /opt/ros)` (ROS source.sh method), and load the libraries corresponding to it
- Stop shipping ROS shared libraries as there is no real trustworthy method to find the user distribution (previous solution will give the first distribution in alphabetical order if multiple ones are installed), and encourage the user to run webots `webots_ros` and `roslaunch` to use ROS-based controllers

DavidMansolino, 2020-02-28:
We should indeed stop polluting the user `LD_LIBRARY_PATH` path by default for every controllers with these libraries, instead, we should simply add them in a runtime.ini file for the 2 controllers using them.

DavidMansolino, 2020-03-04:
This was fixed in #1426 and will be available from the nightly builds of tomorrow.

closed 2020-03-04 by DavidMansolino