cyberbotics/webots #1424: import of cad-models

pmi2020, 2020-03-03:
Hey,

for a project we'd like to implement cad-files (format e.g. stl or step) to the webots environment.
Is there a possibility to import cad-files or convert them to proto-files?

Thanks!

DavidMansolino, 2020-03-03:
Hi @pmi2020,

I am currently working on a feature that will allow you to directly refer to STL (among other files) from Webots to add Shapes, see #1419 

In the meantime, you can export your robot in VRML from SolidWorks and then import this VRML file in Webots directly (not that this will import only the shapes, you will then have to add solids, bounding objects, physics, sensors, etc.). Or import from SolidWorks to Blender and then Blender to Webots, here is an example with Autodesk Inventor  (Solidworks works similarly): https://youtu.be/L0FVsFD2rS4

If you have any further questions, please use our Discord channel: https://discordapp.com/invite/nTWbN9m

closed 2020-03-03 by DavidMansolino

alirezakarimi92, 2021-11-07:
Hi, 
does webots support from *.stl format files for now?

cktyhub, 2024-05-05:
Hello, I am unable to follow the steps demonstrated in the video: file>>import. The software version I am using is 2023b, and I cannot find the import option in the software.