cyberbotics/webots #1432: Multiple Simulation Reset Slow Down Simulation Speed

DavidMansolino, 2020-03-04:
**Describe the Bug**
When using the `wb_supervisor_simulation_reset` multiple time it takes at each iteration longer to reset the simulation.

**Steps to Reproduce**
1. Open any simulation without controller (to make it simpler)
2. Add a robot node and set it's `supervisor` field to `TRUE`.
3. Assign the following controller to the robot:
```Python
from controller import Supervisor

# create the Robot instance.
supervisor = Supervisor()

timestep = 640
while supervisor.step(timestep) != -1:
    supervisor.simulationReset()
```
4. Run the simulation in fast mode.
5. You will see the average speed decreasing slowly at the beginning and then continue decreasing slowly. Even worse, it looks like the controller runs for longer before resetting the simulation at each step.

**Expected behavior**
Each run should be exactly same speed and length.

DavidMansolino, 2020-04-07:
Now when trying to reproduce this issue, I have an even worse problem, when running this controller in run or fast mode Webots get stuck.

DavidMansolino, 2020-04-22:
It actually not stuck, it's just that reset are so close that we can't see the speedometer changing. But if we add a falling object in the scene we can see moving repeatedly.

DavidMansolino, 2020-04-22:
I checked many different scenarios and I can't unfortunately reproduce this issue anymore (both on Ubuntu and MacOS).

Simon-Steinmann, 2020-05-04:
Is this issue fixed in the newest build? I have stable installed, where this bug still exists and need this to work.

DavidMansolino, 2020-05-04:
I can't reproduce it indeed in the development environment, I would, therefore, suggest trying with the nightly builds to check if you still have problems:
https://github.com/cyberbotics/webots/releases

DavidMansolino, 2020-06-29:
This seems to be fixed (not reproducible anymore).

closed 2020-06-29 by DavidMansolino