cyberbotics/webots #1434: Warning OpenGL in console output when toggling "Show contact points"

remi-cyberbotics, 2020-03-04:
Following the tutorial guide: 
https://www.cyberbotics.com/doc/guide/tutorial-2-modification-of-the-environment?tab-language=c#create-a-ball

It is said at the end of the section that we can toggle View/Optional Rendering/Show Contact Points.
Once the option is checked, *and* the simulation has been run, a looping warning shows up in the console and says:
```
OpenGL error: GL_INVALID_OPERATION
```

**System**
 - Ubuntu 18.04.4 LTS - 5.3.0-40-generic
 - NVIDIA GeForce RTX 2060 6GB 
 - Webots compiled from source

DavidMansolino, 2020-03-27:
@lukicdarkoo has the exact same issue.

lukicdarkoo, 2020-03-27:
The same is happening using `webots_ros2` and Webots compiled from source (branch `develop`, commit 7a1573ca569de6dde14cbce826f0b6c49290d671)

DavidMansolino, 2020-03-31:
I have been able to get more information about this error:
```
GL_INVALID_OPERATION in glMapBufferRange(length = 0)
GL_INVALID_OPERATION in glUnmapBuffer(buffer is not mapped)
```

This is probably raised internally by the WREN dynamic mesh (WrDynamicMesh).

DavidMansolino, 2020-04-01:
This seems to be the problematic call:
https://github.com/cyberbotics/webots/blob/bf600fa908cd2035d6b11f5a80f713f6a1349a67/src/wren/DynamicMesh.cpp#L416

closed 2020-04-01 by DavidMansolino