cyberbotics/webots #1439: Race condition between Supervisor::simulationReset() and Robot::step()

ethanwu10, 2020-03-08:
**Describe the Bug**
When `Robot::step()` / `wb_robot_step()` is called immediately after `Supervisor::simulationReset()` / `wb_supervisor_simulation_reset()`, and Webots is under relatively high CPU load, the step will wait for the simulation time to reach `duration` ms after the time at which the reset was issued, even if the simulation time has been reset to 0s

**Steps to Reproduce**
1. Load the provided controller code into a supervisor node in a resource-intensive scene (I used the Mavic demo on my laptop—which runs at about 0.5x—and was able to reproduce very reliably)
2. Run the simulation
3. Observe the simulation time after the controller pauses the simulation

**Expected behavior**
The simulation should be stopped at 0.032s (should be reproducible when the controller is loaded into a completely empty scene)

**System**
 - Operating System: Arch Linux
 - Graphics Card: Intel Integrated UHD Graphics 620

Controller code:
```c
#include <webots/robot.h>
#include <webots/supervisor.h>
#include <stdio.h>

int main()
{
	wb_robot_init();
	while (wb_robot_step(32) != -1 && wb_robot_get_time() < 5);
	wb_supervisor_simulation_reset();
	printf("Reset\n");
	wb_robot_step(32);
	printf("Time: %f (should be 5.056)\n", wb_robot_get_time());
	wb_robot_step(32);
	printf("Time: %f (should be 0.032)\n", wb_robot_get_time());
	wb_supervisor_simulation_set_mode(WB_SUPERVISOR_SIMULATION_MODE_PAUSE);
	wb_robot_cleanup();
	return 0;
}
```

DavidMansolino, 2020-03-11:
I can indeed reproduce the issue your are mentioning.
However, the issue is visible only if the controller step (i.e. 32 in your example) is bigger than the simulation step (i.e. 8 in the Mavic demo case).
As a temporary workaround, I would suggest to use the same time step for the supervisor controller and the simulation.

DavidMansolino, 2020-03-11:
The issue was fixed in #1448.
Please try with the nightly build from tomorrow: https://github.com/cyberbotics/webots/releases
If you still have problems do not hesitate to re-open the issue.

closed 2020-03-11 by DavidMansolino