cyberbotics/webots #1441: Webots Crashing with Python simulationReset()

lordNil, 2020-03-09:
**Describe the Bug**
When the code gets to simulationReset(), the program pauses for a few seconds and then crashes. 
I have a single custom made robot.
I have tried this just by itself, and integrated in my Machine Learning code, both times it crashes

I posted this on Discord, and was told to make a bug report. 

**Steps to Reproduce**
1. Single Robot with Supervisor = True
2. Click on Run simulation
3. Wait until simulationReset() is called. 
4. See the program crash

**Expected behavior**
The simulation should be reset, while the controller is not reset and continues. 

**System**
 - Operating System: Win 10
 - Graphics Card:  NVIDIA GeForce GTX 1650 and Intel 630

Controller code:

from controller import Robot
from controller import Supervisor

robot = Supervisor()
timestep = int(robot.getBasicTimeStep())
max = 100
tim = 0
# Main loop:
while robot.step(timestep) != -1:
    tim = tim+1
    if tim > max:
        robot.simulationReset()
    pass

omichel, 2020-03-10:
I could reproduce this crash and found a workaround: after you set `Robot.supervisor` to TRUE, save the world file and revert it. From there, it will work as expected. But I acknowledge this is a bug and we are going to fix it.

lordNil, 2020-03-10:
Can you clarify what " reverting " the world file is?
I have tried reloading the world file and also loading it again from file after setting robot.supervisor to True.  It still crashes. 

Anyways, thanks for the help.
Some of the main machine learning libraries are in Python (like Tensorflow and Pytorch), and it would be great to get it to work efficiently in Webots.

omichel, 2020-03-11:
Yes, reverting means reloading the world file. In my case, it doesn't crash when I reload the file and it works as expected... I will double-check this again. Meanwhile, can you download the latest [nightly build](https://github.com/cyberbotics/webots/releases) and let us know if the crash is gone for you as well?

I am re-opening this issue as the hotfix in #1445 is not very user friendly. Ideally, we shouldn't have to save and reload, so that a field value is taken into account.

omichel, 2020-03-11:
I confirm that in my case, when I modify the world file setting the `Robot.supervisor` field to `TRUE`, save it and reload it, Webots doesn't crash and everything works as expected. If that's not the case for you, can you share your world file and controller files so that I can try to reproduce your crash here?

lordNil, 2020-03-13:
This should be my world

omichel, 2020-03-13:
Thank you. I can indeed reproduce the crash. I will investigate...

DavidMansolino, 2020-03-13:
@lordNil the crash you reported is not directly related to the supervisor, I have created a bug report about it here: #1454

DavidMansolino, 2020-03-13:
This other problem was fixed in #1455 and will be available in the nightly of tomorrow.

PR #1445 (2020-03-11): Fixed supervisor crash
Fixes #1441.

This patch will prevent Webots from crashing following the procedure described in #1441.
However, the supervisor will not be working until the world is saved and reloaded.
closed 2020-04-07 by DavidMansolino