cyberbotics/webots #1443: Crash when adding IndexedLineSet node to a Shape's geometry

remi-cyberbotics, 2020-03-10:
Add a new node in the scene tree, (Base nodes -> Shape).

Try to add a new node to its geometry field (Base nodes -> IndexedLineSet), it crashes.

DavidMansolino, 2020-03-11:
This is due to an assertion failure, here is the stack:
```
#0  0x00007f7829261e97 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007f7829263801 in __GI_abort () at abort.c:79
#2  0x00007f782925339a in __assert_fail_base (fmt=0x7f78293da7d8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x55e7cc2c64ea "index >= 0 && index < size()", file=file@entry=0x55e7cc2c652e "vrml/WbMFInt.hpp", line=line@entry=58, function=function@entry=0x55e7cc2c68e0 <WbMFInt::item(int) const::__PRETTY_FUNCTION__> "const int& WbMFInt::item(int) const") at assert.c:92
#3  0x00007f7829253412 in __GI___assert_fail (assertion=0x55e7cc2c64ea "index >= 0 && index < size()", file=0x55e7cc2c652e "vrml/WbMFInt.hpp", line=58, function=0x55e7cc2c68e0 <WbMFInt::item(int) const::__PRETTY_FUNCTION__> "const int& WbMFInt::item(int) const") at assert.c:101
#4  0x000055e7cbf1fc78 in WbMFInt::item(int) const (this=0x55e7d7850360, index=0) at vrml/WbMFInt.hpp:58
#5  0x000055e7cbf204ad in WbMFIterator<WbMFInt, int>::next() (this=0x7ffe6a9b65d0) at vrml/WbMultipleValue.hpp:78
#6  0x000055e7cbf5e1d6 in WbIndexedLineSet::computeCoordsData(float*) (this=0x55e7d78c1180, data=0x55e7d76c9bd0) at nodes/WbIndexedLineSet.cpp:147
#7  0x000055e7cbf5e0cd in WbIndexedLineSet::buildWrenMesh() (this=0x55e7d78c1180) at nodes/WbIndexedLineSet.cpp:128
#8  0x000055e7cbf5dd30 in WbIndexedLineSet::createWrenObjects() (this=0x55e7d78c1180) at nodes/WbIndexedLineSet.cpp:80
#9  0x000055e7cbea2aed in WbBaseNode::finalize() (this=0x55e7d78c1180)
```

DavidMansolino, 2020-03-11:
The problem is this line:
https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbIndexedLineSet.cpp#L147
The `it.next()` is called before checking if the iterator has a next value, we should first check if it has and if not return 0.
@remi-cyberbotics, if you want, this might be another 'good first issue'.

closed 2020-03-12 by remi-cyberbotics