cyberbotics/webots #1460: Webots' C compiler won't work !?

sxbug, 2020-03-16:
Today, I use webots 2020b nightly build 13 / 3 / 2020 to open the previously available program and recompile it. The result shows an error prompt:

Make clean
Makefile:74: Files/Webots /resources/Makefile.include: No such file or directory
make: *** No rule to make target 'Files/Webots /resources/Makefile.include'. Stop.
Clean finished.

make -j 6
Makefile:74: Files/Webots /resources/Makefile.include: No such file or directory
make: *** No rule to make target 'Files/Webots /resources/Makefile.include'.  Stop.
Nothing to be done for build targets.

Whether a space Extra after 'Files/webots' ?

**System**
 - Operating System:Windows 10 
 - Graphics Card:NVIDIA GeForce GTX 750Ti

DavidMansolino, 2020-03-16:
I just tried with the 2020b nightly build 13 / 3 / 2020 and i am unfortunately not able to reproduce the issue.

Where did you installed Webots?
Which controller are you trying to recompile?

sxbug, 2020-03-16:
I had installed in c:\program files\webots..
Any C Controller .

There was no problem before。
Thank you

sxbug, 2020-03-16:
I found the reason. The Makefile that in the  controllers directory has been changed. Just replace it.

Thanks

DavidMansolino, 2020-03-16:
You're completely right, an update of the make command did break the compilation,
all the Webots makefiles have been fixed accordingly: #1304

omichel, 2020-03-18:
I would assume we can close this issue. If not please, reopen it.

closed 2020-03-18 by omichel