cyberbotics/webots #1481: Add optional timestamps to console messages

remi-cyberbotics, 2020-03-19:
When loading a world the console does not get cleaned automatically, it is not a problem but maybe some users would like to associate timestamps to the output, so it matches the moment when they are working on simulations. It should be a setting to match user's preference.

omichel, 2020-03-20:
You are raising two issues here:

1. clearing the console when loading a new world.
2. displaying a time stamp on every console print out.

For the first one, I would say that it should be implement by allowing a controller to clear the console. I believe it was possible at some point in Webots, but it doesn't seems to be possible any more. I would implement this functionality as a new ANSI control code: `"\033[2J"` which is the standard code to clear the console. We already support ANSI control codes in the Webots console to change background/foreground colors, font weight, etc. See https://github.com/cyberbotics/webots/blob/master/src/webots/gui/WbConsole.cpp#L396-L463

Also some documentation and a demo about the usage of ANSI control codes to change colors, font weight, clear the console, etc. would be welcome.

For the second one, I believe it is better to let the user implement his own `custom_printf(...)` function where she will be free to display a time stamp header to every message using his favorite format. Also, she may want to display simulation time or real time.

remi-cyberbotics, 2020-03-23:
:eyes:

DavidMansolino, 2020-03-23:
> eyes

Just assigning yourself is enough :wink:

remi-cyberbotics, 2020-03-23:
> > eyes
> 
> Just assigning yourself is enough wink

Olivier told me so :sweat_smile:

remi-cyberbotics, 2020-03-23:
So we plan to add a new escape sequence to enhance the capability of the Webots console.

As such, from a controller, when it will send the escape code "\033[2J", it will be read by the console and so associated to a terminal clear action. This way the user won't be bothered to have it cleared when he didn't decide it.

A C header (to start with, and later all supported languages...) will be created in which all those ANSI supported code will be listed, so the user won't have to look for individual ANSI codes if he/she wants to color the console output for example.

The documentation will be improved accordingly to those improvements, with the list of supported codes. Since we'll be there, a world example might be created to reflect all those features.

closed 2020-04-08 by remi-cyberbotics