cyberbotics/webots #1483: ConveyorPlatform PROTO object has trouble to carry objects on its belt and is not used anywhere

remi-cyberbotics, 2020-03-20:
The ConveyorPlatform PROTO object does not carry elements as it should be, instead they go through the belt and their behavior is undefined. Objects might get carried out after a while or stay stuck on/in the belt.

omichel, 2020-03-20:
Also a simulation world using this ConveyorPlatform PROTO should be added.

DavidMansolino, 2020-03-20:
I have tried and for me it is working fine (with the Mac mini), but I have the feeling that the problem might come from the bounding objects. 
The bounding object of the track itself is extremely thin, the box might touch the bounding box beneath the track (which is not part of the track itself), we should try to make the box bounding object beneath the track part of the track bounding object, this looks safer.

PR #1497 (2020-03-27): Attempt to Fix ConveyorPlatform
**Description**
Attempt to fix #1483

@remi-cyberbotics since I am not able to reproduce the issue, can you please check if this fixes the issue?
closed 2020-03-27 by DavidMansolino