cyberbotics/webots #1488: How to use camera if using "universal_robots_ros" contaoller?

ipa-rwu, 2020-03-23:
Hi everyone,

I am setting up environment with UR5e and camera by using ROS. I check documents and I found there is a camera example in webots_ros. But for ur5e I need to use "universal_robots_ros". How to combine them?

kind regards,
Ruichao

DavidMansolino, 2020-03-23:
Hi @ipa-rwu, the simplest solution is to add a `Robot` node in the ``toolSlot`` field of the ur5e robot, then you can add the ``Camera`` node in the ``children`` field of this new robot and assign the ``ros`` controller to this new robot.
Alternatively, if you want to use only one robot/controller you have to extend the ros controller of the ur5e robot to handle a camera.

ipa-rwu, 2020-03-23:
@DavidMansolino Thank you for your fast response. I added `camera node` to  `toolSlot` now. But I am confused about how to control this node, for example as you said: "assign the ros controller to this new robot". Shall I modify the code `universal_robots_ros`?

DavidMansolino, 2020-03-23:
You are missing an intermediate Robot node to which you have to assign the 'ros' controller, something like:
```
Ur5e {
  toolSlot Robot {
    children [
      Camera {
      }
    ]
    controller "ros"
  }
}
```

ipa-rwu, 2020-03-23:
@DavidMansolino If using this way,  is it that the camera is attached to the end effector?

omichel, 2020-03-23:
Yes, the `toolSlot` is located at the end effector.

ipa-rwu, 2020-03-23:
@DavidMansolino @omichel I see, then the camera will move with the arm. Is it any solution to make it static. I want to set up a simulation environment that the camera is mounted on top of work space and it is static.

omichel, 2020-03-23:
You mean not attached to the robot at all? In such a case, you should create the Robot node at the root level in the scene tree.

ipa-rwu, 2020-03-23:
@omichel yes, the camera won't attach UR5e at all. Then I will have two controllers that one is `universal_robots_ros` and the other one is `ros`. How can I run these two together? I tried to create a robot node with a camera in the root level. But when I run it I just saw controller `universal_robots_ros` is running.

DavidMansolino, 2020-03-24:
Did you set the `controller` field of your second robot to `ros` ?

ipa-rwu, 2020-03-24:
@DavidMansolino @omichel Thank you very much. I can use camera now. I will close this issue.

closed 2020-03-24 by ipa-rwu