cyberbotics/webots #1493: Input Directly RGB values and not HEX values in LED colour change

RDaneelOlivav, 2020-03-25:
Hi,
I'm trying to set the color of led based on the camera readings RGB colors. But I cant manage to find any conversion tool in Webots API that allos to convert RGB readings form 

image_array = camera_object.getImageArray();
# image_array = [[[116, 55, 35]], [[117, 57, 37]], [[115, 57, 39]]]
Into the HEX format that LED requires ( R8G8B8 ). 

Any idea on how this could be accomplished?
Isnt there a :

led_camera.set(value=[[116, 55, 35])?

DavidMansolino, 2020-03-25:
Hi @RDaneelOlivav this is more a programming question than a Webots question, which language are you using?

Please do not hesitate to use the Webots Discord channel for question: https://discordapp.com/invite/nTWbN9m

closed 2020-03-26 by omichel

RDaneelOlivav, 2020-03-28:
I'm using Python
Thanks True, I'll use the discord ;) Thanks for the tip.