cyberbotics/webots #1496: Web interface: jQuery UI dialog cannot be moved and resized on iOS

stefaniapedrazzi, 2020-03-25:
**Describe the Bug**
jQuery UI dialogs cannot be moved or resized from a iOS device.
I tested on a iPad with two different browsers (Safari and Brave) running robotbenchmark.net.

**Steps to Reproduce**
1. Go to robotbenchmark.net and start a benchmark
2. Try to move or resize the info window
-> nothing happens

All the other actions works as expected.

This was working when the mobile interface has been developed.
So it should be due to some changes in iOS/browser.
On Android it works as expected and in Linux/Chrome simulating any iPhone/iPad device it works too.

omichel, 2020-03-25:
Did you try to update to the latest version of jQuery?

stefaniapedrazzi, 2020-03-25:
No, I didn't try but the touch events for jQuery UI are handled by the touch punch library that has not been updated since 5 years and we are alrady using the latest version (0.2.3)

stefaniapedrazzi, 2020-04-01:
I tested on the newly release iOS 13.4, but I still get the same bug.

stefaniapedrazzi, 2021-05-26:
To be checked again with the new wrenjs version.

omichel, 2022-06-13:
We are not using jQueryUI any more.

ThomasOliverKimble, 2022-06-20:
We just tested the new floating windows on iOS and they are some other issues that need to be dealt with: they are not even showing up in the first place. I will assign myself to this issue and try and tackle it today.

ThomasOliverKimble, 2022-06-21:
The new floating windows have been tested and work well on iOS. Closing the issue!

closed 2022-06-21 by ThomasOliverKimble