cyberbotics/webots #1503: Added Samples for every Devices

DavidMansolino, 2020-03-27:
**Describe the Bug**
According to discussion in #1483, it would be nice to have at least one controller + world for every device PROTO distributed within Webots.

Here are some missing ones (non exhaustive list):
  - ConveyorPlatform (missing world).
  - MultiSenseS21 (missing controller and world).
  - Velodyne (missing controller and world).

ygoumaz, 2022-11-08:
Currently, I believe all devices contained in `projects/devices` don't have any sample world or controller. Is it still relevant?

omichel, 2022-11-08:
It is difficult to find out in which world(s) these devices are actually used. Maybe we could add the list of worlds using a specific device in the XML proto list which is generated by a python script. That would allow us to display this information on webots.cloud in the PROTO viewer so that users can quickly access a sample simulation that uses this device.