cyberbotics/webots #1506: Update System Requirements Doc

DavidMansolino, 2020-03-27:
**Describe the Bug**
The documented supported OS in the system requirement are widely outdated:
https://cyberbotics.com/doc/guide/system-requirements

The Mac part is 3 years in the past and for Windows we can at least remove Windows 7 as it is today in the end of life state.

remi-cyberbotics, 2020-04-03:
So should we support what the companies are still supporting as OSes ?

omichel, 2020-04-03:
We should not support officially unsupported OSes.
But we should not necessarily support all the officially supported OSes.
We should in any case support the more recent ones.

remi-cyberbotics, 2020-04-06:
So I am proposing the following list:

Microsoft:
Supported: Windows 10, Windows 8.1
Not supported: All version below including 32 bits variants. (Microsoft has stopped support for Windows 8)

Apple:
Supported: 10.15 Catalina, 10.14 Mojave, 10.13 High Sierra
Unsupported: Versions below.

In all case, I suggest we mention it might be possible to run webots on a not-too-old unsupported version without guarantying it.

DavidMansolino, 2020-04-06:
Looks good to me!

omichel, 2020-04-06:
I would remove Mojave (we use to support only two versions of mac OS).
Also, I would remove the "Not supported" line which doesn't help much.
Anyhow, please go ahead with a PR, that's easier to review than here.

closed 2020-04-07 by remi-cyberbotics