cyberbotics/webots #1516: Specify enabled interactions with the scene in WBPROJ

stefaniapedrazzi, 2020-04-01:
The interactions with the 3D scene can be split into 4 different groups:
1. move viewpoint (translate, rotate, zoom and tilt)
2. select objects and show context menu
3. move objects (translate, rotate, lift with handles or SHIFT key)
4. apply force and torque to an object

In the View menu we already have the options `Lock Viewpoint` and  `Disable Selection` that blocks actions of group 1 and 2. These options are already stored in the WBPROJ file.

I would propose to move the existing items into a submenu called `Scene interactions` and add two additional items:
* "Disable Moving Objects" -> "objectMoveDisabled" in WBPROJ
* "Disable Applying Force and Torque" -> "forceAndTorqueDisable" in WBPROJ

omichel, 2020-04-03:
That's very good. But in addition to this, we might want to disable more things, like scene tree editing, controller editing, console output, etc.

stefaniapedrazzi, 2020-04-15:
Ok, so for the moment I will also add the disabling of the fast mode.

But if we plan later to add options to disable any kind of actions, then it is probably better to move these options to a dedicated tab in the Preferences.

closed 2020-06-11 by stefaniapedrazzi