cyberbotics/webots #1520: JPEG textures not displayed, generate warning about "Unsupported image format"

telramund, 2020-04-04:
**Describe the Bug**

This only happens on Intel HD Graphics 630 GPU. Another system with NVIDIA GeForce 1050Ti does not have this problem. JPEG textures are not displayed, the following warnings are generated:

WARNING: Texture image size of '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_base_color.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_base_color.jpg': Unsupported image format.
WARNING: Texture image size of '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_roughness.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_roughness.jpg': Unsupported image format.
WARNING: Texture image size of '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_normal.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_normal.jpg': Unsupported image format.
WARNING: Texture image size of '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_occlusion.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture '/home/smbdy/webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_occlusion.jpg': Unsupported image format.

**Steps to Reproduce**

1. Start Webots
2. Open Hexapod sample world (or any other)

**Expected behavior**

Textures should be displayed and no warnings generated

**System**
 - Webots R2020a revision 1 (most recent)
 - Operating System Gentoo Linux with 4.9.97, on a Thinkpad t470p
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: Intel HD Graphics 630 GPU

**Additional context**
Please do not recommend switching to the NVIDIA GPU that this laptop also has. This is not currently possible for reasons unrelated to Webots.

DavidMansolino, 2020-04-06:
It seems that your system has a conflicting version of libjpeg install which is preventing Webots from loading JPEG images.
Can you give us the output of the following command:
```
ldd $WEBOTS_HOME/lib/webots/qt/plugins/imageformats/libqjpeg.so
```

telramund, 2020-04-06:
linux-vdso.so.1 (0x00007ffcf5e57000)
	libQt5Gui.so.5 => /usr/lib64/libQt5Gui.so.5 (0x00007f1e9993a000)
	libQt5Core.so.5 => /usr/lib64/libQt5Core.so.5 (0x00007f1e99384000)
	libjpeg.so.8 => not found
	libstdc++.so.6 => /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/libstdc++.so.6 (0x00007f1e9910f000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f1e98f40000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f1e98f1b000)
	libGL.so.1 => /usr/lib64/libGL.so.1 (0x00007f1e98e95000)
	libpng16.so.16 => /usr/lib64/libpng16.so.16 (0x00007f1e98e5f000)
	libz.so.1 => /lib64/libz.so.1 (0x00007f1e98e45000)
	libharfbuzz.so.0 => /usr/lib64/libharfbuzz.so.0 (0x00007f1e98d40000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f1e98bf5000)
	libgcc_s.so.1 => /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/libgcc_s.so.1 (0x00007f1e98bd9000)
	libdouble-conversion.so.3 => /usr/lib64/libdouble-conversion.so.3 (0x00007f1e98bc2000)
	libpcre2-16.so.0 => /usr/lib64/libpcre2-16.so.0 (0x00007f1e98b41000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f1e98b3b000)
	libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007f1e98a1b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1e9a117000)
	libGLdispatch.so.0 => /usr/lib64/libGLdispatch.so.0 (0x00007f1e98961000)
	libGLX.so.0 => /usr/lib64/libGLX.so.0 (0x00007f1e9892d000)
	libfreetype.so.6 => /usr/lib64/libfreetype.so.6 (0x00007f1e98872000)
	libgraphite2.so.3 => /usr/lib64/libgraphite2.so.3 (0x00007f1e9884d000)
	libpcre.so.1 => /lib64/libpcre.so.1 (0x00007f1e987da000)
	libX11.so.6 => /usr/lib64/libX11.so.6 (0x00007f1e98696000)
	libbz2.so.1 => /lib64/libbz2.so.1 (0x00007f1e98683000)
	libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00007f1e98659000)
	libXau.so.6 => /usr/lib64/libXau.so.6 (0x00007f1e98654000)
	libXdmcp.so.6 => /usr/lib64/libXdmcp.so.6 (0x00007f1e9864c000)
	libbsd.so.0 => /usr/lib64/libbsd.so.0 (0x00007f1e98633000)

telramund, 2020-04-06:
I also tried symlinking libjpeg.so to libjpeg.so.8. This got rid of the "not found" message in ldd output, but did not fix the issue and triggered another ldd error:
/home/smbdy/webots/lib/webots/qt/plugins/imageformats/libqjpeg.so: /usr/lib64/libjpeg.so.8: version `LIBJPEG_8.0' not found (required by /home/smbdy/webots/lib/webots/qt/plugins/imageformats/libqjpeg.so)
	linux-vdso.so.1 (0x00007fff5a788000)
	libQt5Gui.so.5 => /usr/lib64/libQt5Gui.so.5 (0x00007f7bf07b7000)
	libQt5Core.so.5 => /usr/lib64/libQt5Core.so.5 (0x00007f7bf0201000)
	libjpeg.so.8 => /usr/lib64/libjpeg.so.8 (0x00007f7bf017d000)
	libstdc++.so.6 => /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/libstdc++.so.6 (0x00007f7beff08000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f7befd39000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f7befd14000)
	libGL.so.1 => /usr/lib64/libGL.so.1 (0x00007f7befc8e000)
	libpng16.so.16 => /usr/lib64/libpng16.so.16 (0x00007f7befc58000)
	libz.so.1 => /lib64/libz.so.1 (0x00007f7befc3e000)
	libharfbuzz.so.0 => /usr/lib64/libharfbuzz.so.0 (0x00007f7befb39000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f7bef9ee000)
	libgcc_s.so.1 => /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/libgcc_s.so.1 (0x00007f7bef9d2000)
	libdouble-conversion.so.3 => /usr/lib64/libdouble-conversion.so.3 (0x00007f7bef9bb000)
	libpcre2-16.so.0 => /usr/lib64/libpcre2-16.so.0 (0x00007f7bef93a000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f7bef934000)
	libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007f7bef814000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7bf0f94000)
	libGLdispatch.so.0 => /usr/lib64/libGLdispatch.so.0 (0x00007f7bef75a000)
	libGLX.so.0 => /usr/lib64/libGLX.so.0 (0x00007f7bef726000)
	libfreetype.so.6 => /usr/lib64/libfreetype.so.6 (0x00007f7bef66b000)
	libgraphite2.so.3 => /usr/lib64/libgraphite2.so.3 (0x00007f7bef646000)
	libpcre.so.1 => /lib64/libpcre.so.1 (0x00007f7bef5d3000)
	libX11.so.6 => /usr/lib64/libX11.so.6 (0x00007f7bef48f000)
	libbz2.so.1 => /lib64/libbz2.so.1 (0x00007f7bef47c000)
	libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00007f7bef452000)
	libXau.so.6 => /usr/lib64/libXau.so.6 (0x00007f7bef44d000)
	libXdmcp.so.6 => /usr/lib64/libXdmcp.so.6 (0x00007f7bef445000)
	libbsd.so.0 => /usr/lib64/libbsd.so.0 (0x00007f7bef42c000)

DavidMansolino, 2020-04-06:
It seems indeed ``libjpeg.so.8`` is missing.
How did you install Webots?
For non-Ubuntu OS, we do recommend to use the snap package: https://cyberbotics.com/doc/guide/installation-procedure#installing-the-snap-package

In your case, to fix this problem, you probably simply need to install libjpeg-8, I am not really familiar with Gentoo, but it seems this should do the job:
```
emerge media-libs/jpeg:8
```

telramund, 2020-04-06:
Thanks, once you pointed me in the right direction, I figured it out:
echo "media-libs/jpeg:8 ~amd64" >> /etc/portage/package.keywords
emerge media-libs/jpeg:8

I installed Webots from a compressed tar archive, I believe.

closed 2020-04-06 by telramund

DavidMansolino, 2020-04-06:
Perfect, thank you for the feedback 😃

MaxGaukler, 2020-04-17:
The same issue occurs when using the tarball on Debian 10. (I can't use the deb package in my specific use case for administrative reasons, but probably that package wouldn't work either.). Unpacking libjpeg.so.8.1.2 from Ubuntu 18's libjpeg-turbo8 package https://packages.ubuntu.com/bionic/amd64/libjpeg-turbo8/download to `webots/lib/webots` helps. Debian 10 doesn't have libjpeg8.

Would it be possible to include libjpeg8 in the tarball?

DavidMansolino, 2020-04-20:
@MaxGaukler, yes we can add it to the package indeed.
Have you tried with the snap package? With the snap package it should be included.

MaxGaukler, 2020-04-20:
I can't use the snap package in my use-case because it requires administrative privileges, and some admins consider the snap system as problematic for security -- at least it had some major security issues in the past.

DavidMansolino, 2020-04-20:
@MaxGaukler, libjpeg8 is now distributed with the tarball package (see #1567), this will be available from the nightly build of tomorrow.