cyberbotics/webots #1521: ROS with Webots example gives issue related with bublebee and NVIDIA

RDaneelOlivav, 2020-04-06:
**Describe the Bug**
When launching this command:

export WEBOTS_HOME=/snap/webots/current/usr/share/webots
roslaunch webots_ros e_puck_line.launch

I get the following error:

/usr/bin/primusrun: line 41: warning: command substitution: ignored null byte in input
primus: fatal: Bumblebee daemon reported: error: [XORG] (EE) Failed to load module "nvidia" (module does not exist, 0)

================================================================================REQUIRED process [webots-2] has died!
process has finished cleanly
log file: /home/rdaneel/.ros/log/05423ab0-77dd-11ea-b367-4c1d96a7e63a/webots-2*.log
Initiating shutdown!
================================================================================

I've checked and the nvidia drivers are working, the glxgears tests pass.. So I don't know exactly what's the issue here.

DavidMansolino, 2020-04-06:
To understand if the problem comes from the ROS integration or Webots on your computer, please try launching Webots with this world:
```
webots /snap/webots/current/usr/share/webots/projects/languages/ros/worlds/e-puck_line.wbt
```

RDaneelOlivav, 2020-04-06:
It works perfectly the launch:

webots /snap/webots/current/usr/share/webots/projects/languages/ros/worlds/e-puck_line.wbt

But when I launch it my own catkin_ws, it gives the above mentioned error. 

export WEBOTS_HOME=/snap/webots/current/usr/share/webots
roslaunch webots_ros e_puck_line.launch

**webots_ros_ws/src/webots_tests_pkg/launch/**test_e_puk.launch ****

<?xml version="1.0"?>
<launch>
  
  <arg name="no-gui" default="false," doc="Start Webots with minimal GUI"/>
  <include file="$(find webots_ros)/launch/webots.launch">
    <arg name="mode" value="realtime"/>
    <arg name="no-gui" value="$(arg no-gui)"/>
    <arg name="world" value="/snap/webots/x1/usr/share/webots/projects/languages/ros/worlds/e-puck_line.wbt"/>
  </include>

  <arg name="duration" default="20" doc="Duration in seconds"/>
  <arg name="auto-close" default="false" doc="Startup mode"/>
  <node name="epuck_line" pkg="webots_ros" type="e_puck_line" args="$(arg duration)" required="$(arg auto-close)"/>
</launch>

DavidMansolino, 2020-04-06:
> It works perfectly the launch

That's reassure it means that Webots works fine with your system/GPU.

Which OS/version are you using? And which version of ROS?

RDaneelOlivav, 2020-04-06:
OS is Ubuntu 18.04
ROS melodic
I installed it through apt:

sudo apt-get install ros-melodic-webots-ros

DavidMansolino, 2020-04-06:
I just tried on Ubuntu 18.04 with the package coming from APT and unfortunately I am not able to reproduce the issue at all, everything is working fine in my machine.
It seems something is going wrong with your graphic setup.
Do you also have this issue with other ROS packages using openGL (such as Rviz for example)?

RDaneelOlivav, 2020-04-06:
No, it only happens in webots with ROS  I'm affraid.
I use ROS a lot at work, locally, different versions... No issue there.

DavidMansolino, 2020-04-07:
It looks like Webots doesn't like to be started from the launch file on your computer.
May I ask you to try tree things?

1. Can you start the sample without the launch file, this means:
   1. Starting roscore
   2. Starting Webots manually with: ``webots /snap/webots/current/usr/share/webots/projects/languages/ros/worlds/e-puck_line.wbt``
   3. Starting the node: ``rosrun webots_ros e_puck_line`
2. Before starting the launch file, make sure to export WEBOTS_HOME: ``export WEBOTS_HOME=/snap/webots/current/usr/share/webots``.
3. make sure to start Webots with your NVIDIA graphic card:
   1. Make sure that your configuration works fine with: ``primusrun glxgears``
   2. Modify the way Webots is started, here is the launcher of Webots: https://github.com/cyberbotics/webots_ros/blob/master/src/webots_launcher.py#L32
I would suggest to change this line to: ``command = [primusrun, os.path.join(os.environ['WEBOTS_HOME'], 'webots'), '--mode=' + options.mode, options.world]``

DavidMansolino, 2020-07-14:
Since we have no news about this issue since several months I am allowing myself to close this issue.

closed 2020-07-14 by DavidMansolino