cyberbotics/webots #1524: The process crashed some time after starting successfully

abylikhsanov, 2020-04-06:
**Describe the Bug**
After trying to go through the tutorial for creating the controller [here](https://cyberbotics.com/doc/guide/tutorial-1-your-first-simulation-in-webots?tab-language=python) I get a warning:

```
WARNING: e_p: The process crashed some time after starting successfully.
WARNING: 'e_p' controller crashed.
```

**Steps to Reproduce**
Follow the tutorial for creating a new controller and use the same code in C++ or Python. The bug seems to happen when creating a `Robot` object.

**System**
 - Operating System: [Linux Ubuntu 18.04]
 - Graphics Card: [NVIDIA GeForce RTX 2070 8 GB.]

DavidMansolino, 2020-04-06:
Can you share the code you are using (just to make 100% sure it is correct according to the tutorial) ?
You are using the 2020a-revision1 version of Webots right?

abylikhsanov, 2020-04-06:
Here is the code:

```
#include <webots/Robot.hpp>

// Added a new include file
#include <webots/Motor.hpp>

#define TIME_STEP 64

// All the webots classes are defined in the "webots" namespace
using namespace webots;

int main(int argc, char **argv) {
 Robot *robot = new Robot();

 // get the motor devices
 Motor *leftMotor = robot->getMotor("left wheel motor");
 Motor *rightMotor = robot->getMotor("right wheel motor");
 // set the target position of the motors
 leftMotor->setPosition(10.0);
 rightMotor->setPosition(10.0);

 while (robot->step(TIME_STEP) != -1);

 delete robot;

 return 0;
```

I installed the binaries (via `apt`) and I do remember it was 2020 version but do not remember letter or revision (I will check tomorrow in the office) but should be the latest release if I installed the binaries.

DavidMansolino, 2020-04-07:
I can indeed reproduce the crash (at least with the Python version of the controller).

abylikhsanov, 2020-04-07:
Hi, yes the version is `R2020a rev 1`

DavidMansolino, 2020-04-07:
I can indeed reproduce it with ``R2020a rev 1`` (but not with my development environment), I am currently installing a nightly build to see if this happens too.

DavidMansolino, 2020-04-07:
I can confirm it is crashing with the nightly releases too.

DavidMansolino, 2020-04-07:
The crash happens only with Python3.x (it is working fine with Python 2.7), and here is the stack on Mac:
```
Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [37599]

VM Regions Near 0x8:
--> 
    __TEXT                 000000010688b000-000000010688c000 [    4K] r-x/r-x SM=COW  /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.python.python             	0x0000000107124eac PyTuple_New + 252
1   org.python.python             	0x00000001071279aa PyType_Ready + 218
2   org.python.python             	0x000000010712796c PyType_Ready + 156
3   _controller.so                	0x0000000106fc925d SwigPyPacked_TypeOnce() + 77
4   _controller.so                	0x0000000106f80baf SwigPyPacked_type() + 47
5   _controller.so                	0x0000000106f7ec62 PyInit__controller + 34
6   org.python.python3            	0x00000001069ad123 _PyImport_LoadDynamicModuleWithSpec + 579
7   org.python.python3            	0x00000001069acb70 _imp_create_dynamic + 352
8   org.python.python3            	0x00000001068b764e _PyMethodDef_RawFastCallDict + 542
9   org.python.python3            	0x00000001068b7148 PyCFunction_Call + 200
10  org.python.python3            	0x0000000106988f65 _PyEval_EvalFrameDefault + 29365
11  org.python.python3            	0x000000010698d108 _PyEval_EvalCodeWithName + 2888
12  org.python.python3            	0x00000001068b6e54 _PyFunction_FastCallKeywords + 228
13  org.python.python3            	0x000000010698c40c call_function + 780
14  org.python.python3            	0x0000000106988b78 _PyEval_EvalFrameDefault + 28360
15  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
16  org.python.python3            	0x000000010698c40c call_function + 780
17  org.python.python3            	0x0000000106988b5b _PyEval_EvalFrameDefault + 28331
18  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
19  org.python.python3            	0x000000010698c40c call_function + 780
20  org.python.python3            	0x0000000106988c18 _PyEval_EvalFrameDefault + 28520
21  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
22  org.python.python3            	0x000000010698c40c call_function + 780
23  org.python.python3            	0x0000000106988c18 _PyEval_EvalFrameDefault + 28520
24  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
25  org.python.python3            	0x000000010698c40c call_function + 780
26  org.python.python3            	0x0000000106988c18 _PyEval_EvalFrameDefault + 28520
27  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
28  org.python.python3            	0x00000001068b8c96 object_vacall + 614
29  org.python.python3            	0x00000001068b8d7d _PyObject_CallMethodIdObjArgs + 173
30  org.python.python3            	0x00000001069abab9 PyImport_ImportModuleLevelObject + 1657
31  org.python.python3            	0x0000000106987303 _PyEval_EvalFrameDefault + 22099
32  org.python.python3            	0x000000010698d108 _PyEval_EvalCodeWithName + 2888
33  org.python.python3            	0x0000000106981c10 PyEval_EvalCode + 48
34  org.python.python3            	0x000000010697f297 builtin_exec + 583
35  org.python.python3            	0x00000001068b764e _PyMethodDef_RawFastCallDict + 542
36  org.python.python3            	0x00000001068b7148 PyCFunction_Call + 200
37  org.python.python3            	0x0000000106988f65 _PyEval_EvalFrameDefault + 29365
38  org.python.python3            	0x000000010698d108 _PyEval_EvalCodeWithName + 2888
39  org.python.python3            	0x00000001068b6e54 _PyFunction_FastCallKeywords + 228
40  org.python.python3            	0x000000010698c40c call_function + 780
41  org.python.python3            	0x0000000106988b78 _PyEval_EvalFrameDefault + 28360
42  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
43  org.python.python3            	0x000000010698c40c call_function + 780
44  org.python.python3            	0x0000000106988b5b _PyEval_EvalFrameDefault + 28331
45  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
46  org.python.python3            	0x000000010698c40c call_function + 780
47  org.python.python3            	0x0000000106988c18 _PyEval_EvalFrameDefault + 28520
48  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
49  org.python.python3            	0x000000010698c40c call_function + 780
50  org.python.python3            	0x0000000106988c18 _PyEval_EvalFrameDefault + 28520
51  org.python.python3            	0x00000001068b737d function_code_fastcall + 237
52  org.python.python3            	0x00000001068b8c96 object_vacall + 614
53  org.python.python3            	0x00000001068b8d7d _PyObject_CallMethodIdObjArgs + 173
54  org.python.python3            	0x00000001069abab9 PyImport_ImportModuleLevelObject + 1657
55  org.python.python3            	0x0000000106987303 _PyEval_EvalFrameDefault + 22099
56  org.python.python3            	0x000000010698d108 _PyEval_EvalCodeWithName + 2888
57  org.python.python3            	0x0000000106981c10 PyEval_EvalCode + 48
58  org.python.python3            	0x00000001069c4b9e PyRun_FileExFlags + 174
[... 43 lines cut ...]
[... cut ...]
    0x7fff682da000 -     0x7fff682e1ffb  libsystem_symptoms.dylib (1238.60.1) <2EDF9CE8-4091-30A8-B125-0F25E579694A> /usr/lib/system/libsystem_symptoms.dylib
    0x7fff682e2000 -     0x7fff682f8ff2  libsystem_trace.dylib (1147.80.3) <AE11B1AC-352A-37F0-BCA9-055609070104> /usr/lib/system/libsystem_trace.dylib
    0x7fff682fa000 -     0x7fff682ffffb  libunwind.dylib (35.4) <E867ACDE-EADE-3C91-A2A4-0C401788FD47> /usr/lib/system/libunwind.dylib
    0x7fff68300000 -     0x7fff68335ffe  libxpc.dylib (1738.80.8) <51E3E807-9133-3605-BB5F-D59ED6404ABF> /usr/lib/system/libxpc.dylib
```

abylikhsanov, 2020-04-07:
How about C++?

DavidMansolino, 2020-04-07:
> How about C++? I can debug but I am not sure that it will be worth to do as you might already know where it crashes

For me the C++ works fine, it is very strange that it is not working for you. Which version of Webots are you using?

abylikhsanov, 2020-04-07:
The same. Actually, when I used Proto file instead of the text editor, it worked

DavidMansolino, 2020-04-08:
I am sorry but I don't understand what you mean by 'using PROTO file' ?

dfremont, 2022-04-12:
I'm getting essentially the same crash with R2022a and Python 3.8 or 3.9 (3.7 works fine). A Python supervisor with the one line `from controller import Supervisor` is enough to trigger the problem. Here's the backtrace with Python 3.8:

```
Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [14152]

VM Regions Near 0x8:
--> 
    __TEXT                      104477000-10447b000    [   16K] r-x/r-x SM=COW  /usr/local/Cellar/python@3.8/3.8.13/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.python.python             	0x0000000104bb08e1 PyTuple_New + 241
1   org.python.python             	0x0000000104bb3559 PyType_Ready + 233
2   org.python.python             	0x0000000104bb3515 PyType_Ready + 165
3   _controller.so                	0x0000000104a2a34f SwigPyPacked_TypeOnce() + 143
4   _controller.so                	0x00000001049c7e0f SwigPyPacked_type() + 47
5   _controller.so                	0x00000001049c5e92 PyInit__controller + 34
6   org.python.python             	0x000000010455b432 _PyImport_LoadDynamicModuleWithSpec + 632
7   org.python.python             	0x000000010455adb5 _imp_create_dynamic + 291
8   org.python.python             	0x00000001044ce7f1 cfunction_vectorcall_FASTCALL + 169
9   org.python.python             	0x00000001044a0333 PyVectorcall_Call + 108
10  org.python.python             	0x000000010453daeb _PyEval_EvalFrameDefault + 28754
11  org.python.python             	0x0000000104540d09 _PyEval_EvalCodeWithName + 1952
12  org.python.python             	0x00000001044a0a51 _PyFunction_Vectorcall + 227
13  org.python.python             	0x00000001045401e6 call_function + 346
14  org.python.python             	0x000000010453d5c9 _PyEval_EvalFrameDefault + 27440
15  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
16  org.python.python             	0x00000001045401e6 call_function + 346
17  org.python.python             	0x000000010453d5a9 _PyEval_EvalFrameDefault + 27408
18  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
19  org.python.python             	0x00000001045401e6 call_function + 346
20  org.python.python             	0x000000010453d671 _PyEval_EvalFrameDefault + 27608
21  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
22  org.python.python             	0x00000001045401e6 call_function + 346
23  org.python.python             	0x000000010453d671 _PyEval_EvalFrameDefault + 27608
24  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
25  org.python.python             	0x00000001045401e6 call_function + 346
26  org.python.python             	0x000000010453d671 _PyEval_EvalFrameDefault + 27608
27  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
28  org.python.python             	0x00000001044a1d37 object_vacall + 277
29  org.python.python             	0x00000001044a1e78 _PyObject_CallMethodIdObjArgs + 217
30  org.python.python             	0x0000000104559ef9 PyImport_ImportModuleLevelObject + 1564
31  org.python.python             	0x000000010453bc95 _PyEval_EvalFrameDefault + 20988
32  org.python.python             	0x0000000104540d09 _PyEval_EvalCodeWithName + 1952
33  org.python.python             	0x00000001045369f6 PyEval_EvalCode + 51
34  org.python.python             	0x00000001045342b6 builtin_exec + 592
35  org.python.python             	0x00000001044ce7f1 cfunction_vectorcall_FASTCALL + 169
36  org.python.python             	0x00000001044a0333 PyVectorcall_Call + 108
37  org.python.python             	0x000000010453daeb _PyEval_EvalFrameDefault + 28754
38  org.python.python             	0x0000000104540d09 _PyEval_EvalCodeWithName + 1952
39  org.python.python             	0x00000001044a0a51 _PyFunction_Vectorcall + 227
40  org.python.python             	0x00000001045401e6 call_function + 346
41  org.python.python             	0x000000010453d5c9 _PyEval_EvalFrameDefault + 27440
42  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
43  org.python.python             	0x00000001045401e6 call_function + 346
44  org.python.python             	0x000000010453d5a9 _PyEval_EvalFrameDefault + 27408
45  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
46  org.python.python             	0x00000001045401e6 call_function + 346
47  org.python.python             	0x000000010453d671 _PyEval_EvalFrameDefault + 27608
48  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
49  org.python.python             	0x00000001045401e6 call_function + 346
50  org.python.python             	0x000000010453d671 _PyEval_EvalFrameDefault + 27608
51  org.python.python             	0x00000001044a08ec function_code_fastcall + 106
52  org.python.python             	0x00000001044a1d37 object_vacall + 277
53  org.python.python             	0x00000001044a1e78 _PyObject_CallMethodIdObjArgs + 217
54  org.python.python             	0x0000000104559ef9 PyImport_ImportModuleLevelObject + 1564
55  org.python.python             	0x000000010453bc95 _PyEval_EvalFrameDefault + 20988
56  org.python.python             	0x0000000104540d09 _PyEval_EvalCodeWithName + 1952
57  org.python.python             	0x00000001045369f6 PyEval_EvalCode + 51
[... cut ...]
59  org.python.python             	0x000000010456dea0 run_mod + 82
60  org.python.python             	0x000000010456dfcd pyrun_file + 133
61  org.python.python             	0x000000010456c785 PyRun_SimpleFileExFlags + 655
62  org.python.python             	0x0000000104583ed1 Py_RunMain + 1891
63  org.python.python             	0x0000000104584218 pymain_main + 306
64  org.python.python             	0x0000000104584266 Py_BytesMain + 42
65  libdyld.dylib                 	0x00007fff204daf3d start + 1```

BenjaminDeleze, 2022-04-25:
Hi,
What is your MacOS version?
Are you using Python from Homebrew or Python.org?

dfremont, 2022-04-25:
I'm using macOS 11.6.2. The log I posted above was with Python 3.8 installed through Homebrew, but I've gotten a similar crash with Python 3.9 installed from Python.org. I've seen the crash both when running Webots inside and outside of a virtual environment (in all cases with the "Python command" option left as the default "python").

dfremont, 2022-05-19:
Is there any other information I can provide to help fix this? We're OK using Python 3.7 if we have to for the moment, but it would be nice to be able to use newer Python versions.

omichel, 2022-06-20:
I believe this problem is fixed in the new version we are preparing. Can you test the R2022b [nightly build](https://github.com/cyberbotics/webots/releases) and confirm the issue is gone? I tested with Python 3.7, 3.8, 3.9 and 3.10 (from python.org) on the latest version of macOS 11 and everything works as expected. I am closing the issue for now. Feel free to re-open if needed.

dfremont, 2022-06-20:
In the mean time I upgraded to macOS 12.3.1, and it's still crashing for me with Python 3.9 (from python.org):

```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [34540]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4301877232
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100697000-100917000    [ 2560K] r-x/r-x SM=COW  ...R/*/python3.9

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Python                        	       0x1017c78d0 _PyObject_GC_Alloc + 48
1   Python                        	       0x1017c7a65 _PyObject_GC_New + 21
2   Python                        	       0x1016b7053 PyDict_New + 67
3   _controller.so                	       0x1011af60f SWIG_Python_TypeCache() + 47
4   _controller.so                	       0x1011ad965 PyInit__controller + 21
5   python3.9                     	       0x1007f60f8 _PyImport_LoadDynamicModuleWithSpec + 632
6   python3.9                     	       0x1007f5994 _imp_create_dynamic + 308
7   python3.9                     	       0x100722337 cfunction_vectorcall_FASTCALL + 215
8   python3.9                     	       0x1007c7a29 _PyEval_EvalFrameDefault + 28153
9   python3.9                     	       0x1007cb763 _PyEval_EvalCode + 3091
10  python3.9                     	       0x1006e0420 _PyFunction_Vectorcall + 256
11  python3.9                     	       0x1007ca7ab call_function + 411
12  python3.9                     	       0x1007c760e _PyEval_EvalFrameDefault + 27102
13  python3.9                     	       0x1006e0515 function_code_fastcall + 229
14  python3.9                     	       0x1007ca7ab call_function + 411
15  python3.9                     	       0x1007c75f1 _PyEval_EvalFrameDefault + 27073
16  python3.9                     	       0x1006e0515 function_code_fastcall + 229
17  python3.9                     	       0x1007ca7ab call_function + 411
18  python3.9                     	       0x1007c76ad _PyEval_EvalFrameDefault + 27261
19  python3.9                     	       0x1006e0515 function_code_fastcall + 229
20  python3.9                     	       0x1007ca7ab call_function + 411
21  python3.9                     	       0x1007c76ad _PyEval_EvalFrameDefault + 27261
22  python3.9                     	       0x1006e0515 function_code_fastcall + 229
23  python3.9                     	       0x1007ca7ab call_function + 411
24  python3.9                     	       0x1007c76ad _PyEval_EvalFrameDefault + 27261
25  python3.9                     	       0x1006e0515 function_code_fastcall + 229
26  python3.9                     	       0x1006e1927 object_vacall + 471
27  python3.9                     	       0x1006e1b86 _PyObject_CallMethodIdObjArgs + 246
28  python3.9                     	       0x1007f4826 PyImport_ImportModuleLevelObject + 1350
29  python3.9                     	       0x1007c5f03 _PyEval_EvalFrameDefault + 21203
30  python3.9                     	       0x1007cb763 _PyEval_EvalCode + 3091
31  python3.9                     	       0x1007c0b60 PyEval_EvalCode + 64
32  python3.9                     	       0x1007bd9ba builtin_exec + 618
33  python3.9                     	       0x100722337 cfunction_vectorcall_FASTCALL + 215
34  python3.9                     	       0x1007c7a29 _PyEval_EvalFrameDefault + 28153
35  python3.9                     	       0x1007cb763 _PyEval_EvalCode + 3091
36  python3.9                     	       0x1006e0420 _PyFunction_Vectorcall + 256
37  python3.9                     	       0x1007ca7ab call_function + 411
38  python3.9                     	       0x1007c760e _PyEval_EvalFrameDefault + 27102
39  python3.9                     	       0x1006e0515 function_code_fastcall + 229
40  python3.9                     	       0x1007ca7ab call_function + 411
41  python3.9                     	       0x1007c75f1 _PyEval_EvalFrameDefault + 27073
42  python3.9                     	       0x1006e0515 function_code_fastcall + 229
43  python3.9                     	       0x1007ca7ab call_function + 411
44  python3.9                     	       0x1007c76ad _PyEval_EvalFrameDefault + 27261
45  python3.9                     	       0x1006e0515 function_code_fastcall + 229
46  python3.9                     	       0x1007ca7ab call_function + 411
47  python3.9                     	       0x1007c76ad _PyEval_EvalFrameDefault + 27261
48  python3.9                     	       0x1006e0515 function_code_fastcall + 229
49  python3.9                     	       0x1006e1927 object_vacall + 471
50  python3.9                     	       0x1006e1b86 _PyObject_CallMethodIdObjArgs + 246
51  python3.9                     	       0x1007f4826 PyImport_ImportModuleLevelObject + 1350
52  python3.9                     	       0x1007c5f03 _PyEval_EvalFrameDefault + 21203
53  python3.9                     	       0x1007cb763 _PyEval_EvalCode + 3091
54  python3.9                     	       0x1007c0b60 PyEval_EvalCode + 64
55  python3.9                     	       0x10081125d pyrun_file + 333
56  python3.9                     	       0x10080f32e PyRun_SimpleFileExFlags + 734
57  python3.9                     	       0x10082e7e3 Py_RunMain + 2067
58  python3.9                     	       0x10082ed13 pymain_main + 403
59  python3.9                     	       0x10082ed6b Py_BytesMain + 43
60  dyld                          	       0x10799251e start + 462```

dfremont, 2022-06-20:
(To be clear, that crash log is with the nightly version you suggested.)

omichel, 2022-06-21:
I tested again and it worked out for all these Pythons version from python.org:

- python 3.7.9
- python 3.8.10
- python 3.9.13
- python 3.10.5

Can you check that you have exactly the same versions, including the revision number?

dfremont, 2022-06-21:
I installed 3.9.13 from python.org and that worked! It turns out I was wrong in my last post above: the crash was with Python installed through `pyenv`, not from python.org. I installed 3.9.13 with `pyenv` and got the same crash, so I guess the issue is with `pyenv`. Does Webots not support Python being installed that way?

omichel, 2022-06-22:
OK, this reassuring. Webots only supports out-of-the-box the official Python from https://python.org and also the one from homebrew. However, you can recompile the wrapper for your own flavor of Python as explained here: https://www.cyberbotics.com/doc/guide/using-python#use-an-alternative-python-version

I am closing (again) this issue as I believe it is finally resolved.

closed 2022-06-22 by omichel