cyberbotics/webots #1526: Thymio-specific motion bug (spinning stopped)

ctyeong, 2020-04-06:
Given "left wheel speed" << "right wheel speed", I would expect my *Thymio* robot to keep spinning in a direction (specifically counterclockwise). However, it doesn't appear so whereas others like *e-puck* does so. 

You can find an example video here: https://twitter.com/ssuty/status/1246646131523604480
As described, it was when the left and right wheel speeds were constantly .3 and 6.0, respectively.

DavidMansolino, 2020-04-07:
The Thymio robot is a bit different than the e-puck one because it has a low friction ball that is touching the ground on the bottom front of the robot:
[image]

This requires that you set the contact properties of this ball in your world file, see for example: https://github.com/cyberbotics/webots/blob/master/projects/robots/mobsya/thymio/worlds/thymio2.wbt#L2-L12

DavidMansolino, 2020-04-07:
Feel free to re-open the issue if this doesn't solve the problem.

closed 2020-04-07 by DavidMansolino